google-deepmind/mujoco #1817: Unexpected Bouncing Behavior with Hfield Using Margin>0 (Demo provided)

Josh00-Lu, 2024-07-16:
Dear MuJoCo Team, 

I am encountering an unexpected bouncing issue when I replace a plane with a flat hfield and set margin=1 in my MuJoCo model. Here’s my XML file:

```xml
<mujoco model="hopper">
  <asset>
    <texture name="skybox" type="skybox" builtin="gradient" rgb1=".2 .3 .4" rgb2="0 0 0" width="800" height="800" mark="random" markrgb="1 1 1"/>
    <texture name="grid" type="2d" builtin="checker" rgb1=".1 .2 .3" rgb2=".2 .3 .4" width="300" height="300" mark="edge" markrgb=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>
    <material name="self" rgba=".7 .5 .3 1"/>
    <hfield name="floor" size="500 500 1 10" nrow="500" ncol="500" />
  </asset>
  <visual>
    <headlight ambient=".4 .4 .4" diffuse=".8 .8 .8" specular="0.1 0.1 0.1"/>
    <map znear=".001"/>
    <quality shadowsize="16384"/>
  </visual>
  <compiler angle="degree" coordinate="local" inertiafromgeom="true"/>
  <default>
    <joint armature="1" damping="1" limited="true"/>
    <geom conaffinity="0" condim="1" margin="0.001" solimp=".8 .8 .01" solref=".02 1" rgba=".7 .5 .3 1"/>
  </default>
  <option integrator="RK4" timestep="0.002"/>
  <worldbody>
    <light cutoff="100" diffuse="1 1 1" dir="0 -0.7 -1.3" directional="true" exponent="1" pos="0 70 130" specular=".1 .1 .1"/>
    <!-- <geom conaffinity="1" condim="3" name="floor" pos="0 0 -1" rgba="1 1 1 1" size="200 200 .125" type="hfield" material="grid" hfield="floor" margin="0"/>  -->
    <geom conaffinity="1" condim="3" name="floor" pos="0 0 -2" rgba="1 1 1 1" size="200 200 .125" type="hfield" material="grid" hfield="floor" margin="1"/>
    <body name="0" pos="0 0 1.25">
      <camera name="track" mode="trackcom" pos="0 -3 1" xyaxes="1 0 0 0 0 1"/>
      <joint type="free"/>
      <geom friction="0.9" fromto="-0 0 0.2 0 0 -0.2" size="0.05" type="capsule"/>
    </body>
  </worldbody>
</mujoco>
```

This strange bouncing always happens with margin > 0 with type `hfield`, but does not happen when the geom is using a `plane` type. Could anyone help diagnose the cause of this issue?

https://github.com/user-attachments/assets/05257a0c-9a26-4060-8892-a9742c8dc356

Operating system: MacOS 14.5
MuJoCo version: test on both v3.1.6 and v3.2.0

Josh00-Lu, 2024-07-18:
Should I decrease the `timestep` of simulator?

yuvaltassa, 2024-07-29:
Positive margins are not well supported in general, we are actively working on a fix.
We'll keep this issue open and close it when the fix is ready.
Thanks for pointing this out.

yuvaltassa, 2026-10-07:
Fixed in 8626d3c

closed 2026-10-07 by yuvaltassa