google-deepmind/mujoco #1818: mj_inverse and mj_rne for calculating the torque of actuator when qacc is zero (statics)

MaabSadig, 2024-07-16:
Hi,

I'm a master student  and I'm trying to use MuJoCo for checking the motor torque values at statics for my model that has two actuators.

I'm looking for some help with mj_inverse and mj_rne functions. At the beginning I tried using mj_inverse but I got very large values of the torques. When switched to mj_rne the values are better but still sometimes it prints zeros for all the values.
I am using RK4, gravity is  0 0 -9.8 , I am defining the inertial parameters for each body and I have one equality connect.

Unites are : kg, m, kg.m^2, m/s^2, so the expected torque unit is N.m

This is the function I defined to calculate the torques at statics using **mj_inverse**:

def compute_gravity_compensation(model, data):
    # Set the desired accelerations to zero
    data.qacc[:] = 0

    # Compute the torques required for zero acceleration using inverse dynamics
 
    mj.mj_inverse(model, data)

    return data.qfrc_inverse

And this to calculate the torques at statics using **mj_rne**:

def compute_gravity_compensation(model, data):
    
    data.qacc[:] = 0
  
    result = np.zeros(model.nv)

    mj.mj_rne(model, data, 0, result)

    return result

yuvaltassa, 2024-07-16:
Stop using RK4 and mj_inverse will work

MaabSadig, 2024-07-17:
I switched to the default solver but still I got large values.
Another question: is there any way to check the Coriolis value? (I mean without including the gravity),  for example in qfrc_bias it contains both (g+C), I want to make sure the output is pure gravitational in statics.

MaabSadig, 2024-07-17:
General question in Mujoco, the variables stored in qfrc_ are all considered forces and will be in force unit?

yuvaltassa, 2024-07-17:
Yes they are in the units appropriate for the dof, force for linear, torque for rotational.

If you make a MRE with an example of what you think are wrong forces I'll be able to help you.

> is there any way to check the Coriolis value? (I mean without including the gravity), for example in qfrc_bias it contains both (g+C), I want to make sure the output is pure gravitational in statics.

Not really since gravity is added inside mj_rne. One workaround would be disable gravity and manually add it back in `xfrc_applied`

closed 2024-07-17 by yuvaltassa

MaabSadig, 2024-07-18:
So int this code, when I use mj_rne I got zero values in qfrc_bias, and when I use mj_inverse I got very large values in qfrc_inverse. 

def compute_gravity_compensation(model, data):
    # Set the desired accelerations to zero
    data.qacc[:] = 0

    result = np.zeros(model.nv)

    mj.mj_rne(model, data, 0, result)
 
    return result

for i in range(N):
    for j in range(N):
        data.qpos[:] = 0
        data.qvel[:] = 0

        data.qpos[2] =  q2[j] #q2
        data.qpos[3] =  q1[i]  #q1
        gravity_comp = compute_gravity_compensation(model, data)

        data.ctrl[0] = gravity_comp[3] #actuator 1
        data.ctrl[1] =  gravity_comp[2] #actuator 2

        mj.mj_forward(model,data)
        
        # Torque force to be compared with  the analytical ones
        m_q1[i][j] = data.qfrc_actuator[3] 
        m_q2[i][j] = data.qfrc_actuator[2]
       
Would you explain what do you mean by setting the values of gravity manually in xfrc_applied?

yuvaltassa, 2024-07-18:
I mean disable gravity and set `xfrc_applied[body, 2] = -m*g` to apply gravity there.

Presumably the forces you are seeing come from constraints or something like this. You can enable the [fwdinv](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag-fwdinv) flag that will run the inverse dynamics and verify that it returns the correct thing.