google-deepmind/mujoco #1819: Editing mjSpec struct and calling the mj_compile interface again, it will return a null pointer.

xyims, 2024-07-16:
**I have modified the main function in the sample/base.cc file, and now it looks like this:**

```
// ...original code...

mjSpec* spec = nullptr;

// ...original code...

int main(int argc, const char** argv) {
    spec = mj_parseXML("../../model/humanoid/humanoid.xml", nullptr, error, 1000);

    (void) mj_compile(spec, nullptr);  // compile spec to model, but ignore the return value

    // https://mujoco.readthedocs.io/en/3.2.0/programming/modeledit.html#model-elements
    mjsBody* world   = mjs_findBody(spec, "world"); 
    mjsGeom* my_geom = mjs_addGeom(world, nullptr);
    my_geom->type    = mjGEOM_BOX;
    my_geom->size[0] = my_geom->size[1] = my_geom->size[2] = 0.5;

    m = mj_compile(spec, nullptr); // compile again
    std::printf("mjModel* == %p\n", m); // return a null pointer

// ...original code...
}

// ...original code...
```
- the second call of mj_compile(spec, nullptr) will return a null pointer.

- In addition, if I use mj_recompile interface after normally calling the mj_compile interface, MuJoCo will prompt an ERROR: Could not allocate memory.

- my solution: Comment out line 3139 of src/user/user_model.cc. Then everything will be in working order.:
```
      if (list[i]->id!=-1 && list[i]->id!=i) {
         // throw mjCError(list[i], "incompatible id in %s array, position %d", mju_type2Str(type), i);
      }
```

  - Operating system: windows11 22631.3880.
  - MuJoCo version: 3.2.0.

I'm not sure if this is a bug, or if I just haven't found the right way to use it.

quagla, 2024-07-16:
Why are you compiling the first time? You can edit the spec directly after parsing.

If you are getting a null pointer from compilation, you can call mjs_getError to get a more informative error about what went wrong.

yuvaltassa, 2024-07-16:
@quagla we should support mj_recompile with NULL mjData

commit 2664cc13 (2024-07-17): Allow passing a null pointer in mj_recompile.

Fixes #1819.

PiperOrigin-RevId: 653242587
Change-Id: I36dbd6ed54abdf21abeb62cf41ee88f159008d01
closed 2024-07-17 by copybara-service[bot]