google-deepmind/mujoco #1820: Combining Models

tudorjnu, 2024-07-16:
Hello!

I'm working on making my model modular to easily interchange components without affecting each other. Essentially, I want to create a base model using separate `xml` files and then combine them in `Python`. While a simple approach would be to “include” them into a `scene`, I want to retain the ability to manipulate these components later.

To be more specific, I have a `scene.xml` file, a decomposed mesh generated using `obj2mjcf`, and a tool defined in a separate `xml`. My goal is to apply attributes like color to the mesh (manipulate the DOM), set parameters (e.g., timestep), and ultimately combine everything into a single model. I attempted a workaround using a `pymjcf` way, but I believe there's a more efficient method. Below is the approach I used to merge the components:

```python
def merge_xml(
    scene_xml_path: Path,
    guidewire_xml_path: Path,
    phantom_xml_path: Path,
) -> str:
    with open(scene_xml_path, "r") as f:
        scene_xml = f.read()
    with open(guidewire_xml_path, "r") as f:
        guidewire_xml = f.read()
    with open(phantom_xml_path, "r") as f:
        phantom_xml = f.read()

    scene_tree = etree.fromstring(scene_xml)
    guidewire_tree = etree.fromstring(guidewire_xml)
    phantom_tree = etree.fromstring(phantom_xml)

    scene_worldbody = scene_tree.find("worldbody")
    guidewire_worldbody = guidewire_tree.find("worldbody")
    phantom_worldbody = phantom_tree.find("worldbody")
    phantom_defaults = phantom_tree.find("default")
    phantom_assets = phantom_tree.find("asset")

    if (
        scene_worldbody is None
        or guidewire_worldbody is None
        or phantom_worldbody is None
    ):
        raise ValueError("All XMLs must contain a worldbody element")

    for element in guidewire_worldbody:
        scene_worldbody.append(element)
    for element in phantom_worldbody:
        scene_worldbody.append(element)

    scene_defaults = scene_tree.find("default")
    if scene_defaults is None:
        scene_defaults = etree.SubElement(scene_tree, "default")
    for element in phantom_defaults:
        scene_defaults.append(element)

    scene_assets = scene_tree.find("asset")
    if scene_assets is None:
        scene_assets = etree.SubElement(scene_tree, "asset")
    for element in phantom_assets:
        scene_assets.append(element)

    compiler = etree.Element("compiler")
    compiler.set("meshdir", phantom_xml_path.parent.as_posix())
    scene_tree.append(compiler)

    merged_xml = etree.tostring(
        scene_tree, pretty_print=True, xml_declaration=True, encoding="UTF-8"
    )
    return merged_xml.decode("utf-8")
```

I would greatly appreciate any advice or suggestions on how to improve this approach. Thank you!

yuvaltassa, 2024-07-16:
https://mujoco.readthedocs.io/en/latest/programming/modeledit.html

https://mujoco.readthedocs.io/en/latest/python.html#model-editing

closed 2024-07-16 by yuvaltassa

tudorjnu, 2024-07-16:
I tried going about it like this:
```python
import cathsim.utils as cathsim_utils
import mujoco

scene_path = cathsim_utils.guidewire_paths.parent / "scene.xml"
model2 = cathsim_utils.guidewire_paths / "guidewire.xml"

spec = mujoco.MjSpec()
spec.from_file(scene_path.as_posix())

model2_spec = mujoco.MjSpec()
model2_spec.from_file(model2.as_posix())

worldbody = spec.worldbody
model2_worldbody = model2_spec.worldbody
model2_body = model2_spec.find_body("guidewire")

test_body = worldbody.add_body()
test_body = model2_body

model = spec.compile()
print(spec.to_xml())
```
I was looking into https://github.com/google-deepmind/mujoco/blob/8e827d0d50d047cea15de5a764597ddf330358a2/python/mujoco/specs.cc to see if I can find something that allows me to map a body to another body from different specs. Is this something that can be achieved?

yuvaltassa, 2024-07-17:
https://mujoco.readthedocs.io/en/latest/programming/modeledit.html#attachment

https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mjs-attachbody