google-deepmind/mujoco #1824: `mjs_firstElement` has a segmentation fault if the spec does not have the element

JewelryForge, 2024-07-18:
Hi,
  I think the proper behavior of `mjs_firstElement` when the spec does not have the element is to return a `nullptr`. Now `mjs_firstElement` (exactly `mjCModel::NextObject`) assumes that the element vectors (e.g. `actuators_`) are not empty and directly accesses the first item.
https://github.com/google-deepmind/mujoco/blob/dd9dd295badd9d09184d9c3c0e7d9641c85dec6a/src/user/user_model.cc#L716
  I think it's better that the function first checks that the vector is not empty and returns a `nullptr` if it's empty.
  The following code can reproduce this issue:
```c++
  auto *spec = mj_makeSpec();
  mjs_firstElement(spec, mjtObj::mjOBJ_HFIELD);
```
  Thank you for your help!

JewelryForge, 2024-07-18:
The same problem also exists in `mjs_firstChild` (i.e. `mjCBody::NextChild`)
In addition, it seems that there are no tests about `mjs_firstElement` currently.

closed 2024-07-18 by copybara-service[bot]

yuvaltassa, 2024-07-19:
Thanks for reporting this!

Please help us find additional issues :slightly_smiling_face:

commit c198a69b (2024-07-18): Return null pointer from mjs_firstChild and mjs_firstElement when the requested list is empty.

Fixes #1824.

PiperOrigin-RevId: 653758711
Change-Id: I6ede762cafd88dcbb0dfc9c2999b7aee24bd2e2a