google-deepmind/mujoco #1830: Touch Grid Sensor Installation

abhihjoshi, 2024-07-22:
Hello, I am currently looking into the grid touch sensor plugin. However, I am a little confused on how to install this plugin. I was wondering if someone could walk me through the steps or point me to the necessary documentation to install this plugin? Also, would it be possible for someone to include how the to get readings from the sensor using the Python Bindings?

abhihjoshi, 2024-07-30:
I think it automatically works in Ubuntu, but not for Mac. Not sure if this was intended behavior?

yuvaltassa, 2024-07-31:
Are the comments in #313 relevant for your problem?

nimrod-gileadi, 2024-08-21:
Can you explain under what circumstance you were encountering this?

i.e. was it:

- A C++ or Python program?
- Were you building MuJoCo libraries from source or downloading the DMG?
- Were you building your own Python bindings? If so, what did you pass for MUJOCO_PLUGIN_PATH?

abhihjoshi, 2024-08-25:
> A C++ or Python program?

This is in a Python program

> Were you building MuJoCo libraries from source or downloading the DMG?

I was building from source

> Were you building your own Python bindings? If so, what did you pass for MUJOCO_PLUGIN_PATH?

Yes, I built the Python bindings myself. I passed in the same as the MUJOCO_PATH. I imagine that we would need to pass in a different path maybe? Would this just be the path to the plugin directory under the MuJoCo root directory?