google-deepmind/mujoco #1834: meshdir does not work for flexcomp while including

xkiixkii, 2024-07-22:
Hi, 

I was trying to include an object into a scene, and I found the compiler cannot find the mesh correctly when I was trying to load the scene that includes the object with its absolute path, however, when I tried to load the scene with its relative path or load the object file which include the scene, it worked pretty fine.

**Here is the test directory structure I have**
```XML
.
├── meshes
│   └── insole.msh
├── model_include_scene.xml
├── model.xml
├── scene_include_model.xml
├── scene.xml
└── test.py

```

**Here are the xml files I used for test**

<details>
  <summary>model.xml</summary>

```XML
<mujoco>
	<compiler meshdir="meshes/" />
	<extension>
		<plugin plugin="mujoco.elasticity.solid" />
	</extension>
	<worldbody>
		<body name="insole_fixed" pos=".12 -.05 .462">
			<flexcomp name="insole" mass="10" dim="3" type="gmsh" file="insole.msh" rgba=".1 .9 .1 1">
				<edge equality="true" />
				<plugin plugin="mujoco.elasticity.solid">
					<config key="young" value="8e5" />
					<config key="poisson" value=".2" />
					<config key="damping" value=".005" />
				</plugin>
			</flexcomp>
		</body>
	</worldbody>
</mujoco>

```

</details>

<details>
  <summary>scene.xml</summary>

```XML
<mujoco>
    <statistic extent="0.3" center="0.3 0 0" />
	<visual>
		<rgba haze="0.15 0.25 0.35 1" />
		<quality shadowsize="8192" />
		<global azimuth="220" elevation="-30" />
	</visual>
	<asset>
		<texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072" />
		<texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300" />
		<material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2" />
	</asset>
    <worldbody>
        <light pos="0.3 0 1.5" dir="0 0 -1" directional="true" />
		<geom name="floor" pos="0 0 0" size="0 0 0.05" type="plane" material="groundplane" />
		<geom name="platform" pos=".65 0 .15" size=".4 .2 .3" type="box" rgba="1 0 0 1" />
    </worldbody>
</mujoco>

```

</details>

<details>
  <summary>model_include_scene.xml</summary>

```XML
<mujoco>
	<include file="scene.xml"/>
	<compiler meshdir="meshes/" />
	<extension>
		<plugin plugin="mujoco.elasticity.solid" />
	</extension>
	<worldbody>
		<body name="insole_fixed" pos=".12 -.05 .462">
			<flexcomp name="insole" mass="10" dim="3" type="gmsh" file="insole.msh" rgba=".1 .9 .1 1">
				<edge equality="true" />
				<plugin plugin="mujoco.elasticity.solid">
					<config key="young" value="8e5" />
					<config key="poisson" value=".2" />
					<config key="damping" value=".005" />
				</plugin>
			</flexcomp>
		</body>
	</worldbody>
</mujoco>

```

</details>

<details>
  <summary>scene_include_model.xml</summary>

```XML
<mujoco>
    <include file="model.xml"/>
    <statistic extent="0.3" center="0.3 0 0" />
	<visual>
		<rgba haze="0.15 0.25 0.35 1" />
		<quality shadowsize="8192" />
		<global azimuth="220" elevation="-30" />
	</visual>
	<asset>
		<texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072" />
		<texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300" />
		<material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2" />
	</asset>
    <worldbody>
        <light pos="0.3 0 1.5" dir="0 0 -1" directional="true" />
		<geom name="floor" pos="0 0 0" size="0 0 0.05" type="plane" material="groundplane" />
		<geom name="platform" pos=".65 0 .15" size=".4 .2 .3" type="box" rgba="1 0 0 1" />
    </worldbody>
</mujoco>

```

</details>

**Here is the test code**

```python
from os import path
import mujoco

dir_absolute = path.dirname(__file__)
relative_scene_include_model = "scene_include_model.xml"
relative_model_include_scene = "model_include_scene.xml"
absolute_scene_include_model = path.join(dir_absolute, relative_scene_include_model)
absolute_model_include_scene = path.join(dir_absolute, relative_model_include_scene)

if __name__ == "__main__":
    try:
        model = mujoco.MjModel.from_xml_path(absolute_scene_include_model)
    except Exception as e:
        print("*" * 50)
        print("scene include model absolute failed")
        print("*" * 50)
        print(e)

    try:
        model = mujoco.MjModel.from_xml_path(relative_scene_include_model)
    except Exception as e:
        print("*" * 50)
        print("scene include model relative failed")
        print("*" * 50)
        print(e)

    try:
        model = mujoco.MjModel.from_xml_path(absolute_model_include_scene)
    except Exception as e:
        print("*" * 50)
        print("model include scene absolute failed")
        print("*" * 50)
        print(e)

    try:
        model = mujoco.MjModel.from_xml_path(relative_model_include_scene)
    except Exception as e:
        print("*" * 50)
        print("model include scene relative failed")
        print("*" * 50)
        print(e)

```

**Issues Encountered:**
```
**************************************************
scene include model absolute failed
**************************************************
XML Error: Error: resource not found via provider or OS filesystem: '/home/kaixing/assets/insole.msh'
Element 'flexcomp', line 0
```

**MuJoCo Version: 3.1.6
OS: Ubuntu 22.04.4 x86_64
Using mujoco python binding**

Josh00-Lu, 2024-10-21:
Same Problem!

quagla, 2025-08-27:
Is this an issue only for flex?
By the way, you can add assets using an asset dict https://github.com/google-deepmind/mujoco/blob/main/python/mujoco/specs_test.py#L777

Neal006, 2026-11-14:
PR #3899 is up.
It fixes this issue.
please review it and let me know if any further changes are required.
Thanks!!