google-deepmind/mujoco #1837: Simulation Trouble when Loading MJCF (Sidebars go black)

Acr0l, 2024-07-22:
Hi,

I recently switched to Windows (10) downloaded MuJoCo v2.3.0, and unzipped it into C:/mujoco. When I executed simulate.exe, the display was correct, however, when loading a model from `mujoco/model` (like `car.xml`, `active_adhesion.xml`, and `balloons.xml`) the sidebar buttons (both left and right) disappear visually (see attachment).

[image]

[image]

(Using `model/car/car.xml`)

The issue is purely on display, as I am able to click on the buttons (which I don't see) and the desired actions still happen. I have tested with both `simulate.exe` and running on the cmd `python -m mujoco.viewer` and `python -m mujoco.viewer --mjcf=model/car/car.xml`.

Attachment with buttons pressed:
[image]

yuvaltassa, 2024-07-23:
You didn't provide the details of your system, but I suspect you have an AMD GPU?

yuvaltassa, 2024-07-23:
Regardless of GPU, there was another GUI-related bug that was fixed recently. Could you please check with the latest version?

Acr0l, 2024-07-24:
> You didn't provide the details of your system, but I suspect you have an AMD GPU?

Yes, I have an AMD Radeon RX 570, I forgot to add the specs. I checked the drivers and opened `simulate.exe` on compatibility mode (ran with Windows 7 and 8), yet it made no difference.

> Regardless of GPU, there was another GUI-related bug that was fixed recently. Could you please check with the latest version?

I tried with the latest (prebuilt) release and (prebuilt) v3.1.6, but neither of them worked. I will shortly try with source code.

yuvaltassa, 2024-07-29:
I believe this is a duplicate of #639 (so closing this one), there is a longstanding issue with AMD GPUs...

@emotodorov has recently hit this issue himself so might take a look at some point :slightly_smiling_face: