google-deepmind/mujoco #1839: Optimizing Collision Detection for Large Terrain Maps in MuJoCo Simulations

Josh00-Lu, 2024-07-23:
Dear MuJoCo Team,

I am currently working on terrain simulations using MuJoCo. I have encountered performance issues primarily due to the size of the terrain maps involved. There are two main factors contributing to the slowdown: 
1. An increase in the number of columns (`ncol`) and rows (`nrow`) of the terrain, which seems to directly affect the simulation speed.
2. The `size` of the terrain `geom` elements also inferences the simulation speed.

I am seeking advice on how to optimize collision detection to only consider the terrain surrounding the current robot, thereby ignoring other parts of the map that are not in immediate proximity to the robot. This would ideally help in reducing the computational load and increase the simulation speed.

Could you please suggest any settings or methods within MuJoCo that would allow for such selective collision detection? Or are there other optimization techniques you would recommend for handling large terrain maps in simulations?

Thank you for your assistance.

yuvaltassa, 2024-07-23:
I think we still haven't implemented BVH midphase for hfields, right @quagla ?

Could you please open a bug for this, perhaps this would be a good task for @thowell under your guidance?

Josh00-Lu, 2024-07-23:
> I think we still haven't implemented BVH midphase for hfields, right @quagla ?
> 
> Could you please open a bug for this, perhaps this would be a good task for @thowell under your guidance?

Sure. I have opened a bug issue for this. Thanks for your prompt reply.