google-deepmind/mujoco #1842: Gradient Propagation from End-Effector Contact Forces to Joint Angles

binbinbai, 2024-07-23:
Hi,I am a student currently studying robotic hands and I have a question regarding gradient propagation.Is there a way to propagate the gradient of the end-effector contact forces to the gradient of joint angles? I understand that the contact forces at the end-effector can be related to joint torques through the Jacobian matrix. However, I am curious about how joint torques can be related to joint angles.
Any guidance or references on this topic would be greatly appreciated! Thank you!

Balint-H, 2024-07-23:
I think for that kind of approach you need to make heavy assumptions about your model due to redundancy (many potential joint configurations can result in the same contact forces). 

E.g. if you know that the only forces acting on your system are the contact force and passive joint forces (or stiffness based actuation), and can assume you reached steady state (no joint velocity/acceleration), then you can try to find a least squares solution to joint angle deviation across your kinematic chain.

binbinbai, 2024-07-24:
thank you very much! @Balint-H

closed 2024-07-24 by binbinbai