google-deepmind/mujoco #1844: How to render objects metallic, preferably by editting XML files?

barikata1984, 2024-07-23:
Hi,

I'm a student and I'm trying to use MuJoCo for rigid body dynamics simulation and computer vision techniques.

I'm looking for some help with how to render an object more metallic. Below is the figure of my teapot model whose body is glass and lid is stainless: 
[image]
As you can see, the stainless lid appears more like plastic or ceramic. How can I make it appear more metallic?

Below is the XML:
```xml
<mujoco model="target">
  <compiler meshdir="assets" texturedir="assets" autolimits="true"/>
  <default>
    <default class="visual">
      <geom group="2" type="mesh" contype="0" conaffinity="0"/>
    </default>
    <default class="collision">
      <geom group="3" type="mesh"/>
    </default>
  </default>
  <asset>
    <material name="236,241,239" specular="1.0" shininess="1.0" rgba="0.925490 0.945098 0.937255 0.220000"/>
    <material name="190,188,186" specular="0.7372549999999999" shininess="1.0" rgba="0.745098 0.737255 0.729412 1.000000"/>
    <mesh file="object_0.obj"/>
    <mesh file="object_1.obj"/>
  </asset>
  <worldbody>
    <body name="object">
      <geom mesh="object_0" material="190,188,186" class="visual"/>
      <geom mesh="object_1" material="236,241,239" class="visual"/>
      <geom mesh="object_0" class="collision"/>
      <geom mesh="object_1" class="collision"/>
    </body>
  </worldbody>
</mujoco>
```
`object_1` corresponds to the lid. 

Besides, I have noticed that there is a description of the `metallic` parameter `asset/material` elements ([link](https://mujoco.readthedocs.io/en/stable/XMLreference.html#asset-material-metallic)). But I need to find out which external renderer supports this parameter. 

The best-case scenario for me is making the lid more metallic only by editing my XML, but information about renderers supporting metallic surface rendering is also helpful.

Thanks in advance!

Balint-H, 2024-07-24:
Hello,

Meshes don't support reflectance which is key in achieving a metallic effect (among other things)

If I understand correctly, explicit compatibility with external renderers will receive further support and guidelines in the near future. Right now, if you are looking for Phyiscally Based Rendering of realistic materials on MuJoCo objects, some of the approaches include 
- Performing the rendering in Blender based un the MuJoCo scene (example: https://arxiv.org/pdf/2404.15194v1)
- Using the Unity plugin of MuJoCo

barikata1984, 2024-07-24:
Hi, @Balint-H,

Thanks for your response. I will look into the paper. 

l'm just curious. In my understanding, mujoco uses opengl, which should have the capability to simulate reflectance. Why has the reflectance not been supported so far?

closed 2024-07-25 by quagla