google-deepmind/mujoco #1845: I am curious about the cause of Unity mujoco crash.

mg638, 2024-07-24:
hello. After installing the mujoco plugin in Unity, I was able to load the model successfully, but a crash occurred while playing, the engine shut down, and I was able to check the following log. Could you possibly predict what could be the cause using this log?

=================================================================
	Managed Stacktrace:
=================================================================
	  at <unknown> <0xffffffff>
	  at Mujoco.MujocoLib:mj_defaultVFS <0x000db>
	  at Mujoco.MjVfs:.ctor <0x0017a>
	  at Mujoco.MjEngineTool:LoadModelFromString <0x000ba>
	  at Mujoco.MjScene:CompileScene <0x000da>
	  at Mujoco.MjScene:CreateScene <0x00d32>
	  at Mujoco.MjScene:Start <0x000a2>
	  at System.Object:runtime_invoke_void__this__ <0x00187>
=================================================================

quagla, 2024-07-24:
Can you try loading the model in simulate and see if you still get the crash?

mg638, 2024-07-24:
It works fine in the simulator.

mg638, 2024-07-24:
The crash phenomenon has disappeared. The cause seems to be that the crash occurred because the simulator version was not set to 3.2.0 when applying version 3.2.0 this time.