google-deepmind/mujoco #1846: Crash when calling mjs_firstElement or mjs_firstChild if no objects of said type are defined

juan-g-bonilla, 2024-07-24:
In the example blelow, calling `mjs_firstElement` or `mjs_firstChild` will crash the program because the spec has no text or actuators, and the body has no sites. This crash also happens for other object types, like joints.

```
#include "mujoco/mujoco.h"
#include <iostream>

char error[1000];

int main(void) {
    auto xml = R"(
        <mujoco>
          <worldbody>
            <body name="foo">
              <joint type="free"/>
            </body>
          </worldbody>
        </mujoco>
    )";

    // load model from file and check for errors
    auto s = mj_parseXMLString(xml, NULL, error, 1000);
    if (!s) {
        std::cerr << error << std::endl;
        return 1;
    }

    //mjs_addText(s);
    //mjs_addActuator(s, NULL);
    //mjs_addSite(mjs_findBody(s, "foo"), NULL);

    // Any of these lines will cause a crash, unless the 
    // corresponding add functions have been called above
    mjs_firstElement(s, mjOBJ_TEXT);
    mjs_firstElement(s, mjOBJ_ACTUATOR);
    mjs_firstChild(mjs_findBody(s, "foo"), mjOBJ_SITE);

    std::cout << "Finished!" << std::endl;
    return 0;
}
```

Operating system:  Windows 11 Pro; Version 23H2; OS build 22631.3880
MuJoCo version: 3.2.0
Compiled using:
```
~> cl src\test.cpp /Iinclude /Iinclude\mujoco /link /LIBPATH:lib mujoco.lib /out:test.exe
Microsoft (R) C/C++ Optimizing Compiler Version 19.34.31937 for x64
Microsoft (R) Incremental Linker Version 14.34.31937.0
```

Thank you all for the fantastic work!

quagla, 2024-07-24:
I think this is the same as https://github.com/google-deepmind/mujoco/issues/1822 which was fixed in https://github.com/google-deepmind/mujoco/commit/e79efa60d6afd191610f5462dc9a9b47aac98278

quagla, 2024-07-24:
I think this is the same as https://github.com/google-deepmind/mujoco/issues/1824 which was fixed in https://github.com/google-deepmind/mujoco/commit/c198a69bbf7eb53ed856a0ae9c2475e1bd603ff2 so I'll close this, if you can compile from source and still have the crash please open another issue :)

closed 2024-07-24 by quagla