google-deepmind/mujoco #1847: Flexcomp with solid plugin using a mesh

tudorjnu, 2024-07-24:
I am experimenting with the creation of a soft model from a mesh. I've followed previous discussions and tips and turned a simple mesh (“bunny.obj”) into a `gmsh` using [fTetWild](https://github.com/wildmeshing/fTetWild). Furthermore, I also simplified the mesh to have 495 faces, as to make it easier to debug. 

When I run the simulation, it gives me the following error:
```
WARNING: Nan, Inf or huge value in QACC at DOF 0. The simulation is unstable. Time = 0.0000.
```

I've tried changing the time step, playing with the parameters and so on, but nothing seems to work.

Below is the minimal example:
<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="bunny">
  <option timestep="0.001"></option>
  <extension>
    <plugin plugin="mujoco.elasticity.solid"></plugin>
  </extension>
  <worldbody>
    <body name="bunny" pos="0 0 1">
      <flexcomp dim="3" file="bunny_simple_.msh" mass="5" name="bunny" type="gmsh">
        <contact condim="1" selfcollide="none" solimp=".95 .99 .0001" solref="0.01 1"></contact>
        <edge damping="1"></edge>
        <plugin plugin="mujoco.elasticity.solid">
          <config key="young" value="5e4"></config>
          <config key="poisson" value="0.2"></config>
        </plugin>
      </flexcomp>
    </body>
  </worldbody>
</mujoco>
```

</details>

I've attached both the model before transformation `bunny.obj` and after making it a `gmsh` here: [bunny_simple.zip](https://github.com/user-attachments/files/16360996/bunny_simple.zip)

Is there any suggestion to get the model running?

tudorjnu, 2024-07-24:
Yes. It seems that it crashes after loading. The whole viewer seems to crash with it.

quagla, 2024-07-24:
Can you try disabling self collisions and playing with the radius attribute of flexcomp? I just want to make sure the initial configuration doesn't contain overlapping sphere at vertices.

tudorjnu, 2024-07-25:
Making the radius `0.001` worked. Thanks!

closed 2024-07-25 by tudorjnu