google-deepmind/mujoco #1848: Error accessing variables in mjSpec

vikashplus, 2024-07-24:
I'm facing problems accessing variables via mjSpec after building Python binding from the latest main. These work in the latest release, but not on the latest main.

```python 
import mujoco

XML=r"""
<mujoco>
  <worldbody>
    <geom type="plane" size=".1 .1 .1"/>
    <body pos="0 0 .1">
      <geom type="sphere" size=".05"/>
    </body>
  </worldbody>
</mujoco>
"""

spec = mujoco.MjSpec()
spec.from_string(XML)
model = spec.compile()

# I can't access these 
print(spec.worldbody.first_body().pos)
print(spec.worldbody.first_body().name)
print(spec.modelname)

body = spec.worldbody.first_body()
site = body.add_site()

#I can't access these
print(site.name,  site.pos)
```

quagla, 2024-07-24:
I have just tried in our internal repo and the test passes

```
Ran 1 test in 0.017s

OK
[0.  0.  0.1]

MuJoCo Model
 [0. 0. 0.]
```

Are you able to run any of these tests? 

https://github.com/google-deepmind/mujoco/blob/5ac5cfb618c11ff4aa0199b25c3856da3d194703/python/mujoco/specs_test.py

vikashplus, 2024-07-24:
Here is the output of the tests 
```bash
Running tests under Python 3.10.14: /opt/homebrew/Caskroom/miniconda/base/envs/myolab_mjx/bin/python
[ RUN      ] SpecsTest.test_assets
[  FAILED  ] SpecsTest.test_assets
[ RUN      ] SpecsTest.test_basic
[  FAILED  ] SpecsTest.test_basic
[ RUN      ] SpecsTest.test_compile_errors_with_line_info
[  FAILED  ] SpecsTest.test_compile_errors_with_line_info
[ RUN      ] SpecsTest.test_element_list
[  FAILED  ] SpecsTest.test_element_list
[ RUN      ] SpecsTest.test_iterators
[  FAILED  ] SpecsTest.test_iterators
[ RUN      ] SpecsTest.test_modelname_default_class
[  FAILED  ] SpecsTest.test_modelname_default_class
[ RUN      ] SpecsTest.test_recompile
[  FAILED  ] SpecsTest.test_recompile
[ RUN      ] SpecsTest.test_uncompiled_spec_cannot_be_written
[       OK ] SpecsTest.test_uncompiled_spec_cannot_be_written
======================================================================
ERROR: test_assets (__main__.SpecsTest)
SpecsTest.test_assets
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/vikashplus/temp/building_mujoco/mujoco/python/dist/../mujoco/specs_test.py", line 264, in test_assets
    mesh.name = 'cube'
AttributeError: 'mujoco._specs.MjsMesh' object has no attribute 'name'

======================================================================
ERROR: test_basic (__main__.SpecsTest)
SpecsTest.test_basic
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/vikashplus/temp/building_mujoco/mujoco/python/dist/../mujoco/specs_test.py", line 37, in test_basic
    self.assertEqual(spec.eulerseq[0], ord('x'))
AttributeError: 'mujoco._specs.MjSpec' object has no attribute 'eulerseq'

======================================================================
ERROR: test_compile_errors_with_line_info (__main__.SpecsTest)
SpecsTest.test_compile_errors_with_line_info
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/vikashplus/temp/building_mujoco/mujoco/python/dist/../mujoco/specs_test.py", line 106, in test_compile_errors_with_line_info
    geom.name = 'MyGeom'
AttributeError: 'mujoco._specs.MjsGeom' object has no attribute 'name'

======================================================================
ERROR: test_element_list (__main__.SpecsTest)
SpecsTest.test_element_list
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/vikashplus/temp/building_mujoco/mujoco/python/dist/../mujoco/specs_test.py", line 229, in test_element_list
    sensor1.name = 'sensor1'
AttributeError: 'mujoco._specs.MjsSensor' object has no attribute 'name'

======================================================================
ERROR: test_iterators (__main__.SpecsTest)
SpecsTest.test_iterators
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/vikashplus/temp/building_mujoco/mujoco/python/dist/../mujoco/specs_test.py", line 242, in test_iterators
    geom1.name = 'geom1'
AttributeError: 'mujoco._specs.MjsGeom' object has no attribute 'name'

======================================================================
ERROR: test_modelname_default_class (__main__.SpecsTest)
SpecsTest.test_modelname_default_class
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/vikashplus/temp/building_mujoco/mujoco/python/dist/../mujoco/specs_test.py", line 164, in test_modelname_default_class
    spec.modelname = 'test'
AttributeError: 'mujoco._specs.MjSpec' object has no attribute 'modelname'

======================================================================
ERROR: test_recompile (__main__.SpecsTest)
SpecsTest.test_recompile
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/vikashplus/temp/building_mujoco/mujoco/python/dist/../mujoco/specs_test.py", line 124, in test_recompile
    geom.size[0] = 1
AttributeError: 'mujoco._specs.MjsGeom' object has no attribute 'size'

----------------------------------------------------------------------
Ran 8 tests in 0.001s

FAILED (errors=7)
```

quagla, 2024-07-24:
That is weird, those tests passed on main

```
specs_test.py::SpecsTest::test_assets PASSED                             [ 94%]
specs_test.py::SpecsTest::test_basic PASSED                              [ 95%]
specs_test.py::SpecsTest::test_compile_errors_with_line_info PASSED      [ 95%]
specs_test.py::SpecsTest::test_element_list PASSED                       [ 96%]
specs_test.py::SpecsTest::test_iterators PASSED                          [ 97%]
specs_test.py::SpecsTest::test_modelname_default_class PASSED            [ 97%]
specs_test.py::SpecsTest::test_recompile PASSED                          [ 98%]
specs_test.py::SpecsTest::test_uncompiled_spec_cannot_be_written PASSED  [ 98%]
```

vikashplus, 2024-07-24:
Let's compare out packages. Here is mine, lets me know if anything stands out. 

```
absl-py==2.1.0
asttokens==2.4.1
certifi==2024.2.2
charset-normalizer==3.3.2
click==8.1.7
cloudpickle==1.2.2
decorator==5.1.1
dm-env==1.6
dm-tree==0.1.8
dm_control==1.0.20
etils==1.7.0
exceptiongroup==1.2.1
executing==2.0.1
flatten-dict==0.4.2
fsspec==2024.3.1
gitdb==4.0.11
GitPython==3.1.43
glfw==2.7.0
gym==0.13.0
h5py==3.11.0
idna==3.7
importlib_resources==6.4.0
ipdb==0.13.13
ipython==8.23.0
jax==0.4.26
jaxlib==0.4.26
jedi==0.19.1
labmaze==1.0.6
lxml==5.2.1
matplotlib-inline==0.1.7
ml-dtypes==0.4.0
mujoco @ file:///Users/vikashplus/temp/building_mujoco/mujoco/python/dist/mujoco-3.2.0.tar.gz#sha256=2937a6fe2972bea7e043a5bfa312f3e56de4f27d066ac3cc94c875d99ef35d74
mujoco-mjx==3.2.0
-e git+ssh://git@github.com/myolab/myo_api.git@c61a7fe5dc7971c7c1358e78e2b9f34f7bc276f3#egg=myo_api
MyoSuite==2.3.0
numpy==1.26.4
opt-einsum==3.3.0
packaging==24.0
parso==0.8.4
pexpect==4.9.0
pillow==10.3.0
pink-noise-rl==2.0.1
prompt-toolkit==3.0.43
protobuf==5.26.1
ptyprocess==0.7.0
pure-eval==0.2.2
pyglet==2.0.15
Pygments==2.17.2
PyOpenGL==3.1.7
pyparsing==3.1.2
requests==2.31.0
scipy==1.13.0
six==1.16.0
sk-video==1.1.10
smmap==5.0.1
stack-data==0.6.3
termcolor==2.4.0
tomli==2.0.1
tqdm==4.66.2
traitlets==5.14.3
trimesh==4.3.1
typing_extensions==4.11.0
urllib3==2.2.1
wcwidth==0.2.13
zipp==3.18.1
```

quagla, 2024-07-24:
It looks like all of your failures come from variables which are defined in an autogenerated file called `spec.scc.inc`. It is autogenerated here

https://github.com/google-deepmind/mujoco/blob/main/python/mujoco/CMakeLists.txt#L412

Is there any way for you to check if perhaps this file is empty during your compilation?

quagla, 2024-07-24:
Closing as @vikashplus solved this by re-running `make_sdist.sh`.

closed 2024-07-24 by quagla