google-deepmind/mujoco #1850: ValueError: XML Error: Error: All nodes must be in single block

federicopilot, 2024-07-24:
Good day.
i am trying to simulate a soft-body in MuJoCo that i have produced with OpenSCAD. My ultimative goal is to create a more complex structure in OpenSCAD and the simulate it in MuJoCo. For testing purpose i am therefore trying to simulate a simple body from OpenSCAD in MuJoCo so that it exhibits flexibile propertes. First i have created an stl file of the object. Since I am using the flexcomp properties I have tried to transfer the file to a GMSH format using the GSMH software exporting it as an msh file (version 4.1 ASCII). I have created a Volume using Geometry->Elementary entities->Add->Volume. With my python Skript and the corresponding msh file i run into the following error: ValueError: XML Error: Error: All nodes must be in single block. I have tried a lot of different strategies and now I am running out of options and desperatley need help to find a possible solution. Thank you very much!

import mujoco as mj
import numpy as np
import os
from mujoco.glfw import glfw

#Define Simulation parameters
sim_time = 5 #Simulation time in s
xml_path = "stl_integraton2.xml"#Name of the full path for the XML file
#Define initial Position
print_camera_config = 1 # set to 1 to print camera config (useful for initializing view of the model)
gmsh_file = "cubic_v4.1.msh"

#MuJoCo model XML
MODEL_XML = f"""

with open(xml_path, "w") as f:
f.write(MODEL_XML)
#...

msh file:
$MeshFormat
4.1 0 8
$EndMeshFormat
$Entities
0 0 1 2
1 0 0 0 5 5 5 0 0
1 0 0 0 5 5 5 0 1 1
2 0 0 0 5 5 5 0 1 1
$EndEntities
$Nodes
3 8 1 8
2 1 1 8
1
2
3
4
5
6
7
8
0 5 5 0 0
5 0 5 0 0
5 5 5 0 0
0 0 5 0 0
0 0 0 0 0
5 5 0 0 0
5 0 0 0 0
0 5 0 0 0
3 1 0 0
3 2 0 0
$EndNodes
$Elements
2 18 1 18
2 1 2 12
1 1 2 3
2 2 1 4
3 6 1 3
4 1 5 4
5 5 1 8
6 1 6 8
7 2 6 3
8 5 2 4
9 2 5 7
10 6 2 7
11 5 6 7
12 6 5 8
3 1 4 6
13 6 2 5 7
14 6 1 5 3
15 5 1 4 2
16 2 6 5 3
17 1 6 5 8
18 5 3 1 2
$EndElements

mohammad200h, 2024-07-24:
Hey, This is a Duplicate issue #1492. You might find explanation in #1724 useful.

1. Short explanation is that when you use GMSH script or app it creates multiple entities.
2. Mujoco accepts single entity hence the error  "All nodes must be in single block"
3. You can use the following script to clean your file so it can be used by Mujoco:
https://github.com/mohammad200h/GMSHConverter/tree/tree

Let me know how it goes :)

mohammad200h, 2024-07-24:
If you look at:
$Nodes
3 8 1 8 
2 1 1 8
you can see 3 8 1 8  which describes a volume 
and 2 1 1 8 which describes a surface.

federicopilot, 2024-07-25:
Thank you @mohammad200h, I followed your advice and was able to generate two files for the surface and volume msh. I Even checked the file in GMSH and saw that the Volume and Surface entities have been seperated which was good. Then i tried to run the binary and Ascii msh file for the Volume and got the following error message: **ValueError: Error: invalid sizes in MSH file 'newcube_volume.msh'.** So far I have tried creating simpler objects and I have tried to use different versions of GMSH file in [format 4.1](https://gmsh.info//doc/texinfo/gmsh.html#MSH-file-format) and [format 2.2](https://gmsh.info//doc/texinfo/gmsh.html#MSH-file-format-version-2-_0028Legacy_0029) (ascii or binary) which i read in the Mujoco Dokumentation. Unfortunately nothing worked and i am  getting the same error code.
i would be very thankful if you could look into it again. 

thank you very much!

here is the original object(i had to change the form slightly but it is very similar to the original object at the begining of the issue)

**newcube.msh:** 
```
$MeshFormat
4.1 0 8
$EndMeshFormat
$Entities
0 0 1 1
1 0 0 0 4 8 3 0 0 
1 0 0 0 4 8 3 0 1 1 
$EndEntities
$Nodes
2 8 1 8
2 1 1 8
1
2
3
4
5
6
7
8
0 8 3 0 0
4 0 3 0 0
4 8 3 0 0
0 0 3 0 0
0 0 0 0 0
4 8 0 0 0
4 0 0 0 0
0 8 0 0 0
3 1 0 0
$EndNodes
$Elements
2 18 1 18
2 1 2 12
1 1 2 3 
2 2 1 4 
3 6 1 3 
4 1 5 4 
5 5 1 8 
6 1 6 8 
7 2 6 3 
8 5 2 4 
9 2 5 7 
10 6 2 7 
11 5 6 7 
12 6 5 8 
3 1 4 6
13 6 5 2 8 
14 1 5 2 4 
15 1 5 8 2 
16 1 6 2 8 
17 6 1 2 3 
18 6 5 7 2 
$EndElements
```

**newcube_volume.msh**
```
$MeshFormat
4.1 0 8
$EndMeshFormat
$Entities
0 0 0 1
1 0 0 0 4 8 3 0 0 
$EndEntities
$Nodes
1 8 1 8
3 1 0 8
1
2
3
4not
5
6
7
8
0 8 3
4 0 3
4 8 3
0 0 3
0 0 0
4 8 0
4 0 0
0 8 0
$EndNodes
$Elements
1 6 1 6
3 1 4 6
1 6 5 2 8 
2 1 5 2 4 
3 1 5 8 2 
4 1 6 2 8 
5 6 1 2 3 
6 6 5 7 2 
$EndElements
```

**newcube_surface.msh**
```
$MeshFormat
4.1 0 8
$EndMeshFormat
$Entities
0 0 1 0
1 0 0 0 4 8 3 0 0 
$EndEntities
$Nodes
1 8 1 8
2 1 1 8
1
2
3
4
5
6
7
8
0 8 3 0 0
4 0 3 0 0
4 8 3 0 0
0 0 3 0 0
0 0 0 0 0
4 8 0 0 0
4 0 0 0 0
0 8 0 0 0
$EndNodes
$Elements
1 12 1 12
2 1 2 12
1 1 2 3 
2 2 1 4 
3 6 1 3 
4 1 5 4 
5 5 1 8 
6 1 6 8 
7 2 6 3 
8 5 2 4 
9 2 5 7 
10 6 2 7 
11 5 6 7 
12 6 5 8 
$EndElements

```
**here is a screenshot of the object:** 
[image]

**here is my code that i am using only showing the most essential parts:** 
`mport mujoco as mj
import numpy as np
import os
from mujoco.glfw import glfw

#Define Simulation parameters
sim_time = 5 #Simulation time in s
xml_path = "stl_integraton2.xml"#Name of the full path for the XML file
#Define initial Position 
print_camera_config = 1 # set to 1 to print camera config (useful for initializing view of the model)
gmsh_file = "newcube_volume.msh"

#MuJoCo model XML
MODEL_XML = f"""
<mujoco>
 
    <asset>
        <mesh name="Box" file="{gmsh_file}" />
    </asset>

    <option timestep="0.01"/>
    <extension>
        <plugin plugin="mujoco.elasticity.solid" />
    </extension>
    <size memory="100M"/>
    
    <worldbody>
        <geom name="floor1" type="plane" pos="0 0 0.3" size="100 100 0.5" euler="0 0 0" rgba="0.9 0.9 0.9 1"/>
        <light diffuse="1 1 1" pos="0 0 100" dir="0 0 -1"/>
        
         
        <flexcomp name="stl_file" type="gmsh" file="{gmsh_file}" dim="3" pos="0 0 10" rgba=".3 .2 .2 1" mass="100">
            <edge equality="true" damping="0.3"/>
            <contact condim="3" solref="0.03 1" solimp=".95 .99 .0001" selfcollide="none"/>
            
            <plugin plugin="mujoco.elasticity.solid">
               <config key="poisson" value="0.2"/> 
               <config key="young" value="1e5"/>
            </plugin>
            
        </flexcomp>

    </worldbody>
</mujoco>
    """

# Save the XML to a file
with open(xml_path, "w") as f:
    f.write(MODEL_XML)
#...

#Configuration values of camera and Simulation environment
# MuJoCo data structures
model = mj.MjModel.from_xml_path(xml_path)  # MuJoCo model
data = mj.MjData(model)  # MuJoCo data
cam = mj.MjvCamera()  # Abstract camera
opt = mj.MjvOption()  # visualization options
...
`

mohammad200h, 2024-07-26:
I tried replicating the issue:
1. I used the initial model you provided and named it mesh_1.msh
```
$MeshFormat
4.1 0 8
$EndMeshFormat
$Entities
0 0 1 2
1 0 0 0 5 5 5 0 0
1 0 0 0 5 5 5 0 1 1
2 0 0 0 5 5 5 0 1 1
$EndEntities
$Nodes
3 8 1 8
2 1 1 8
1
2
3
4
5
6
7
8
0 5 5 0 0
5 0 5 0 0
5 5 5 0 0
0 0 5 0 0
0 0 0 0 0
5 5 0 0 0
5 0 0 0 0
0 5 0 0 0
3 1 0 0
3 2 0 0
$EndNodes
$Elements
2 18 1 18
2 1 2 12
1 1 2 3
2 2 1 4
3 6 1 3
4 1 5 4
5 5 1 8
6 1 6 8
7 2 6 3
8 5 2 4
9 2 5 7
10 6 2 7
11 5 6 7
12 6 5 8
3 1 4 6
13 6 2 5 7
14 6 1 5 3
15 5 1 4 2
16 2 6 5 3
17 1 6 5 8
18 5 3 1 2
$EndElements
```
2. I used   [GMSH cleaner](https://github.com/mohammad200h/GMSHConverter/tree/tree) to clean the model. named mesh_1_out_vol.msh
```
$MeshFormat
4.1 0 8
$EndMeshFormat
$Entities
0 0 0 1
1 0 0 0 5 5 5 0 0 
$EndEntities
$Nodes
1 8 1 8
3 1 0 8
1
2
3
4
5
6
7
8
0 5 5
5 0 5
5 5 5
0 0 5
0 0 0
5 5 0
5 0 0
0 5 0
$EndNodes
$Elements
1 6 1 6
3 1 4 6
1 6 2 5 7 
2 6 1 5 3 
3 5 1 4 2 
4 2 6 5 3 
5 1 6 5 8 
6 5 3 1 2 
$EndElements

```
4. Installed latest [Mujoco simulate](https://github.com/google-deepmind/mujoco/releases)
5.  create model.xml
```
<mujoco model="cube">

  <option timestep="0.01"/>
  <extension>
    <plugin plugin="mujoco.elasticity.solid" />
  </extension>
  <size memory="100M"/>

  <worldbody>
    <geom name="floor1" type="plane" pos="0 0 0.3" size="100 100 0.5" euler="0 0 0" rgba="0.9 0.9 0.9 1"/>
    <light diffuse="1 1 1" pos="0 0 100" dir="0 0 -1"/>
    <!-- Use a body element to contain the mesh geom -->

    <flexcomp name="stl_file" type="gmsh" file="mesh_1_out_vol.msh" dim="3" pos="0 0 10" rgba=".3 .2 .2 1" mass="100">
        <edge equality="true" damping="0.3"/>
        <contact condim="3" solref="0.03 1" solimp=".95 .99 .0001" selfcollide="none"/>

        <plugin plugin="mujoco.elasticity.solid">
           <config key="poisson" value="0.2"/>
           <config key="young" value="1e5"/>
        </plugin>

    </flexcomp>

  </worldbody>

</mujoco>

```
6. Dragged the model into simulate

it loads.

Here is a Zip file: 
[1850.zip](https://github.com/user-attachments/files/16390824/1850.zip)

Here is a screen cast:
https://youtu.be/OgTzOnKUKHw

federicopilot, 2024-07-26:
Hello, 

thank you very much @mohammad200h it appears to work now! I am very thankful since I have been working on this problem for days. I also want to note that MuJoCo only reads the binary GMSH files and not the ASCII format. I am now hoping to simulate self-made, more complex-looking objects, with MuJoCo and make them behave like a non-rigid body (flexible) like in real-life.

mohammad200h, 2024-07-26:
Happy to help:)
Mujoco supports both binary and ASCII.

closed 2024-07-26 by mohammad200h