google-deepmind/mujoco #1851: Setting `fullinertia` to `body/inertia` results in unexpected inertia configuration (X and Z may be inverted?)

barikata1984, 2024-07-24:
Hi, everyone, 

I am facing an issue which may be a bug. I tried to import a model designed by a CAD software and set its inertial params to the model by assigning its mass, CoM, and full inertia, which are output by the CAD, with `dm_control/mjcf1. The assignment itself worked well, but then I noticed the values registered to an MjModel instance were weird. I am not sure, but the data suggests the X and Z axes of the inertial frame treated invertedly.

Below is a code to mimick the issue. The code sets a body's mass, CoM, diagonal inertia, coordinate-transfers the diagonal inertia with a rotation matrix, register the params to the body's model tree, then compares the original params with those computed by mujocod.
```python
from math import radians

import numpy as np
import pandas as pd
from dm_control import mjcf
from mujoco._structs import MjModel
from transforms3d.euler import euler2mat
from transforms3d.quaternions import mat2quat, quat2mat

# register the root body and a body to which iprops are assigned
cf =  mjcf.RootElement()
body = cf.worldbody.add("body", name="test_body")

# body mass
body_mass = 10

# body com
pos_obj_obji = np.array([0, 0, 0])

# orientation of the inertial frame w.r.t the body frame
rx = radians(14)
ry = radians(24)
rz = radians(36)
# rotation from the body's inertial frame to the body frame
# 'bodyi' and 'body' mean the inertial frame and the body frame
rot_body_bodyi = euler2mat(rx, ry, rz, "rxyz")  # intrincic XYZ euler
# quartanion of the inverse
quat_body_bodyi = mat2quat(rot_body_bodyi)

# principal moments of inertia
pixx = 0.04
piyy = 0.05
pizz = 0.06
diaginertia = np.array([pixx, piyy, pizz])
# expanded as a tensor
diaginertia_tensor = np.diag(diaginertia)

# transfer the reference frame to the body frame
inertia_tensor = rot_body_bodyi @ diaginertia_tensor @ rot_body_bodyi.T
ixx = inertia_tensor[0, 0]
iyy = inertia_tensor[1, 1]
izz = inertia_tensor[2, 2]
ixy = inertia_tensor[0, 1]
iyz = inertia_tensor[1, 2]
izx = inertia_tensor[2, 0]
fullinertia = np.array([ixx, iyy, izz, ixy, izx, iyz])

# register the inertial params to the body
body.add("inertial",
         mass=body_mass,
         pos=pos_obj_obji,
         fullinertia=fullinertia,
         )

# spawn model and get mujoco-computed data
m = MjModel.from_xml_string(cf.to_xml_string())
mj_mass = m.body_mass[1]
mj_diaginertia = m.body_inertia[1]
mj_iquat = m.body_iquat[1]

# compare the orignal and mujoco-computed data
cad_gt = [body_mass, *diaginertia, *quat_body_bodyi]
mjc = [mj_mass, *mj_diaginertia, *mj_iquat]
# principal moments of inertia and the inertial frame's orientation w.r.t the body frame
index = ["total_mass", "pixx", "piyy", "pizz", "real", "i", "j", "k"]
print("mass, diaginertia, quat ------------------------------------------------------")
print(pd.DataFrame({"original": cad_gt,
                    "mujoco": mjc,
                    },
                   index=index,
                   ).transpose()
      )
print("")
print("rot_body_bodyi -------------------------------------------------------")
print(f"original:\n{rot_body_bodyi}")
print(f"mujoco:\n{quat2mat(mj_iquat)}")
```
Below is the result:
```
mass, diaginertia, quat ------------------------------------------------------
          total_mass  pixx  piyy  pizz      real         i         j         k
original        10.0  0.04  0.05  0.06  0.915510  0.177141  0.159425  0.324109
mujoco          10.0  0.06  0.05  0.04 -0.354437  0.760094  0.103922  0.534632

rot_body_bodyi ---------------------------------------------------------------
original:
[[ 0.7390738  -0.53696855  0.40673664]
 [ 0.64993158  0.72714855 -0.22100665]
 [-0.17708434  0.42769121  0.88640925]]
mujoco:
[[ 0.40673662  0.53696857  0.7390738 ]
 [-0.22100661 -0.72714856  0.64993158]
 [ 0.88640928 -0.42769117 -0.17708434]]
```
As you can see, the principal moments of inertia for the X and Z axes (pixx and pizz) are inverted. Besides, the basis vectors of the X and Z axes are also inverted between the original orientation of the inertial frame w.r.t the body frame and that computed by mujoco. 

Since I am not proficient in C++, I cannot investigate the corresponding process in the source .cpp thorouoghly. However, I doubt that some processing inside it may handle the X and Z axes invertedly.

Is anybody able to look into it?

My machine setup is
  - Operating system: Ubuntu 22.04
  - MuJoCo version: 3.2.0

Thanks in advance!

quagla, 2024-07-25:
Hi, so the inertia matrix is consistent with the basis. When we do the eigenvalue decomposition, in MuJoCo we reorder them by magnitude. The CAD probably doesn't do that.

barikata1984, 2024-07-30:
Hi, @quagla,

Thanks for the reply. I understood it and checked on paper that the order of eigenvalues does not change the result dynamics computation. 

I am curious. Is that reordering technically advantageous? I just want to know.

yuvaltassa, 2024-07-31:
I don't think there is a specific advantage other than having a known internal convention.

closed 2024-07-31 by yuvaltassa