google-deepmind/mujoco #1856: How to realize wind frictions acting on the Cheetah model?

EmiliyanGospodinov, 2024-07-25:
Hi,

I'm a student and I'm trying to use MuJoCo for training a model-based reinforcement learning agent in different environments. The main goal is to extend model-based reinforcement learning approaches to deal with non-stationary environments, e.g. environments where dynamics(e.g. changing mass, gravity, adding wind friction) and/or reward(e.g. changing target velocity) change over time. Currently, I focus on MuJoCo-based environments from [Gymnasium](https://gymnasium.farama.org/environments/mujoco/)

I'm looking for some help with defining wind frictions that act on the  [Cheetah model](https://github.com/Farama-Foundation/Gymnasium/blob/main/gymnasium/envs/mujoco/half_cheetah_v5.py) My current idea is to add additional actuators that act on certain joints against or in the direction in which the model is moving. 
Below you can see only the part of the Cheetah model that I modified, namely the actuator part. All motors with the gear parameter set are from the original Cheetah model. Additionally, I define one actuator attached to a particular joint with a limited control range between -50 and 50. I exclude the additional actuators from the agent's action space so that the agent does not have control over them (please check the function below), I set their values explicitly during runtime. 

I have two questions: 
1) I'm unsure if I need to set the gear parameter for each additional actuator or not? 
2) Also, according to the documentation, there is no problem with having multiple actuators acting on a single joint, but everything is pretty new to me, and not aware if I'm changing the mode in a way that is still correct?

Here is the modified actuator part:

<details>

```
  <actuator>
    <motor gear="120" joint="bthigh" name="bthigh"/>
    <motor gear="90" joint="bshin" name="bshin"/>
    <motor gear="60" joint="bfoot" name="bfoot"/>
    <motor gear="120" joint="fthigh" name="fthigh"/>
    <motor gear="60" joint="fshin" name="fshin"/>
    <motor gear="30" joint="ffoot" name="ffoot"/>
    <motor ctrllimited="true" ctrlrange="-50 50" joint="rootx" name="frictionrootx"/>
    <motor ctrllimited="true" ctrlrange="-50 50" joint="bthigh" name="frictionbthigh"/>
    <motor ctrllimited="true" ctrlrange="-50 50" joint="bshin" name="frictionbshin"/>
    <motor ctrllimited="true" ctrlrange="-50 50" joint="bfoot" name="frictionbfoot"/>
    <motor ctrllimited="true" ctrlrange="-50 50" joint="fthigh" name="frictionfthigh"/>
    <motor ctrllimited="true" ctrlrange="-50 50" joint="fshin" name="frictionfshin"/>
    <motor ctrllimited="true" ctrlrange="-30 30" joint="ffoot" name="frictionffoot"/>
  </actuator>
```

    def exclude_wind_friction_from_action_space(self):
        """The additional actuators used to realize the wind friction should not be part of the action space of the agent!"""
        bounds = self.model.actuator_ctrlrange.copy().astype(np.float32)[:-3]
        low, high = bounds.T
        self.action_space = Box(low=low, high=high, dtype=np.float32)
</details>

Any feedback/suggestions are welcomed :)

Kallinteris-Andreas, 2024-07-26:
You can use the wind [model option](https://mujoco.readthedocs.io/en/stable/APIreference/APItypes.html#mjoption)
```
>>> import gymnasium
>>> gymnasium.__version__
'1.0.0'
>>> env = gymnasium.make("HalfCheetah-v5")
>>> env.unwrapped.model.opt.wind = ... # add your desired win value here
```

EmiliyanGospodinov, 2024-07-28:
@Kallinteris-Andreas 
I totally overlooked this option, thank you very much for the suggestion I will try that out! 

And just for clarification, if I want to perturbate the forces applied to the joints by introducing additional actuators as discussed above, do I need to set the gear value of each of the new actuators? In this case I would have following definition per additional actuator: 
` <motor gear="50" joint="rootx" name="frictionrootx"/>`. 

According to the documentation [actuator and its parameters](https://mujoco.readthedocs.io/en/stable/XMLreference.html#actuator) the gear will scale the length (and consequently moment arms, velocity and force), but when I don't set the gear value and print all forces, I can still clearly see that the force outputs stored in mjData.actuator_force are changing, e.g. they are the sum of the original and the new actuator per joint. But maybe I overlook something?

Many thanks for the reply!

yuvaltassa, 2024-07-29:
Perhaps you'll want to use [actuatorgroupdisable](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-actuatorgroupdisable)

closed 2024-07-29 by yuvaltassa