google-deepmind/mujoco #1859: [feature request] fisheye camera

ykim104, 2024-07-25:
**Is your feature request related to a problem? Please describe.**
We are using fisheye cameras as visual sensors to teach a robot policy and would like to use Mujoco for generating robot trajectories with visual observations.

**Describe the solution you'd like**
A `fisheye` tag for creating a `body/camera`

**Describe alternatives you've considered**
As an alternative, we would need to try to expand Mujoco's default OpenGL renderer to create a fisheye effect ourselves

**Additional context**

aftersomemath, 2024-07-26:
MuJoCo uses the fixed function OpenGL pipeline. I'm not certain, but I don't think it can be used to render distorted images. Usually a shader would be used for this.

However, it is straightforward render several regular pinhole cameras and create a fisheye image by sampling from the pinhole camera frames.

ykim104, 2024-07-29:
Thank you @aftersomemath for your reply! 

I have another feature request/question. Is there a plan to bring in other OpenGL features like physically-based rendering ([PBR](https://learnopengl.com/PBR/Theory)) or make use of other textures like Normal maps? We believe such improvements in render fidelity could really help the research. Thank you in advance for your feedback!!

yuvaltassa, 2024-07-30:
Yes, we are planning some significant rendering improvements, but these will take some time (end-of-year-ish).
You can already see some early aspects of this as materials can now reference [various kinds of textures](https://mujoco.readthedocs.io/en/latest/XMLreference.html#material-rgb), but these are currently ignored by the native renderer (except for RGB). Closing for now, the rendering improvements, when they are done, will be well-advertised :slightly_smiling_face:

closed 2024-07-30 by yuvaltassa

MarcelTorne, 2025-08-12:
Hi, when I try to render a fisheye camera using different pinhole camera views I am getting a problem that is each one of the cameras seems to have different exposure, the light and render quality of each view is different, which makes the combined fisheye view not usable. Do you have any advice on how to fix that? Thank you!

aftersomemath, 2025-08-12:
Is it possible to provide a small example?

MarcelTorne, 2025-08-13:
Yes, it will take me a bit longer though. Any chance you have examples on your end on how to do this fisheye rendering properly? Thanks again!

aftersomemath, 2025-08-13:
I don't have an example ready to go unfortunately.

A simple way to construct an example is to insert a few cameras at the same position into a scene like humanoid100. Then rotate them so that their field of views "touch". The resulting images can be horizontally stacked to give a strange, but technically correct ultra wide fov image. Any lighting and rendering issues should still be visible.

atyshka, 2025-08-15:
@yuvaltassa One year later, any comments on the roadmap for rendering? I know there has been the development of the madrona renderer, but unfortunately it's rather brittle dependency-wise and not actively maintained. The maintainers seem to be suggesting they're working on a new renderer integrated with Warp?

yuvaltassa, 2025-08-15:
Ya!

We have a working prototype of a completely new rendering pipeline using Filament. It looks amazing.
ETA Q4 25

yuvaltassa, 2025-10-17:
cc @okmatija @haroonq 

It's coming!