google-deepmind/mujoco #1861: Issues with MuJoCo Position Controller

binbinbai, 2024-07-28:
Hello,

I'm currently working on controlling a robotic hand using a position controller in MuJoCo. For the entire motion sequence, I have target joint angles (t_qpos) for each frame. I am using t_qpos as the control signals and passing them to data.ctrl, then calling mj_step to update the data. However, I've noticed significant discrepancies between data.qpos and t_qpos. I realized this was because a single call to mj_step only computes one time step, so I called mj_step() a sufficient number of times. Despite this, some joint angles still exhibit substantial errors (around e-01), while others have minimal errors (e-07).

Is this behavior normal? Are there any methods to improve the accuracy of the joint angles?

Thank you for answering these fundamental questions!

binbinbai, 2024-07-29:
I have discovered that the deviation in the z-direction sliding joint is slightly larger due to the influence of gravity. Can this issue be resolved through gravity compensation?#1393

yuvaltassa, 2024-07-29:
Yes, you can use [body/gravcomp](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-gravcomp)

closed 2024-07-29 by yuvaltassa