google-deepmind/mujoco #1866: spatial Jacobian computation different from textbook (a math intro to robotic manipulation)

MiaoDragon, 2024-07-31:
Hi,

I'm a PhD student using Mujoco for robotic manipulation projects, specifically for trajectory optimization. I'm using Mujoco to compute **the spatial Jacobian**, and which can give **the analytical Jacobian dg/dq**, according to the textbook a mathematical introduction to robotic manipulation page 115 as follows:
[image]

I assumed Mujoco computed the spatial Jacobian in this way. However, when I check the analytical Jacobian computed from Mujoco spatial Jacobian using above equations **against finite differencing analytical Jacobian**, I found that **there is a gap in the translational components**. Hence I wrote my own code to compute the spatial Jacobian according to the textbook as follows:

[image]

I compared the spatial Jacobian $J_{mj}$ by Mujoco and the one $J$ by textbook, and found that the rotational parts are the same, but the translational parts are different.
Then I used the spatial Jacobian by textbook $J$ to obtain the analytical Jacobian $\tilde{J}=\frac{\partial g}{\partial q}$, which is of size $4\times4\times N_v$. I found that the translational part of the analytical Jacobian is the same as the translational part of the mujoco spatial Jacobian $J_{mj}$, as $\tilde{J}[:3,3,:]=J_{mj}[:3,:]$ (I'm using the first 3 rows for translation, and the last 3 rows for rotation).
So I conclude that $\frac{\partial g}{\partial q}[:3,3,:]=J_{mj}[:3,:]$, and $J[3:,:]=J_{mj}[3:,:]$. This means that **the translation part of the mujoco Jacobian is actually the same as the analytical Jacobian translational part; while the rotational part of the mujoco Jacobian is the same as the textbook spatial Jacobian rotational part**.
I think this would cause much confusion due to the mismatch of computations.

Could you explain why the implementation of Jacobian in Mujoco is this way? The proposal is to change the spatial Jacobian to follow the one in the textbook. Please let me know what you think. Thanks!

The code to implement the textbook version of spatial Jacobian can be found below. You can test the Jacobian implementation with the Mujoco one obtained through `mj_jacBody` function. The so3 and se3 files are included in the attached file.

```
"""
specify the robot model for collision checking (self collision) and Jacobian computation.
implemented by mujoco, hpp-fcl, open3d (visualization)
"""

import so3, se3
import mujoco
import hppfcl
import numpy as np
import transformations as tf
import open3d as o3d
import copy
from typing import Union
import numpy.typing as npt
from collections import deque

"""
compute the kinematics trees. The structure is the following:
each node stores the following:
- link name
- link idx
- parent node
- children nodes
- Pi_i exp(xi_i*theta_i) up to the current node
- joints (order as it appears in the XML)
    - joint name
    - joint idx
    - xi_i at rest pose (theta=0)
    - Pi_i exp(xi_i*theta_i) up to the current node, and up to the current joint (since Mujoco joint computation is accoriding to
    the order of the joints in the XML file)
"""

class KinematicsNode:
    class KinematicJoint:
        def __init__(self, joint_idx, link_pose_zeroed, mj_model):
            """
            link_pose_zeroed: the pose of the link when the joint angles are zero
            """
            self.joint_idx = joint_idx
            self.joint_name = mj_model.joint(joint_idx).name
            self.qpos_adr = mj_model.jnt_qposadr[joint_idx]
            self.dof_adr = mj_model.jnt_dofadr[joint_idx]
            # obtain the axis for the joint
            jnt_pos = np.array(mj_model.jnt_pos[joint_idx])
            jnt_pos = link_pose_zeroed[:3,:3].dot(jnt_pos) + link_pose_zeroed[:3,3]
            jnt_axis = np.array(mj_model.jnt_axis[joint_idx])
            jnt_axis = jnt_axis / np.linalg.norm(jnt_axis)
            jnt_axis = link_pose_zeroed[:3,:3].dot(jnt_axis)
            # by default: assume to be hinge joint
            # TODO: add slider joint as well
            xi = se3.screw(jnt_axis, jnt_pos, 1.0, 0.0)  # this is the unit screw
[... 220 lines cut ...]
        node = self.nodes[link_idx]
        print('node.product_exp_xi_theta:', node.product_exp_xi_theta)
        print('node.link_pose_q_zeroed:', node.link_pose_q_zeroed)
        return node.product_exp_xi_theta @ node.link_pose_q_zeroed

    def get_link_spatial_jacobian(self, link_name: str, compute_spatial_jacobian_cols: bool = False,
                                  compute_product_exp_xi_theta: bool = False) -> np.ndarray:
        """
        get the spatial jacobian for the specific link
        """
        if compute_spatial_jacobian_cols:
            self.compute_spatial_jacobian_cols(compute_product_exp_xi_theta)
        cols = np.array(self.spatial_jacobian_cols)
        # * set irrelavent link cols to zero
        # obtain the relavent joint dofs
        link_idx = self.mj_model.body(link_name).id
        node = self.nodes[link_idx]
        relavent_dofs = node.relavent_joint_dofs
        irrelavent_dofs = set(range(self.mj_model.nv)) - set(relavent_dofs)
        irrelavent_dofs = np.array(list(irrelavent_dofs))
        cols[:,irrelavent_dofs] = 0.0
        return cols
```

[so3_se3.zip](https://github.com/user-attachments/files/16435454/so3_se3.zip)

kevinzakka, 2024-08-01:
Hi @MiaoDragon,

MuJoCo computes neither a spatial nor a body Jacobian. If you've used Pinocchio before, MuJoCo computes a Jacobian in `LOCAL_WORLD_ALIGNED`, aka a Jacobian whose frame of reference is centered at the local frame but aligned with the world frame. To obtain a Jacobian expressed in the local frame, aka a body Jacobian, you need to left-multiply by Adjoint[T_fw] where T_fw is the transform of the world frame `w` expressed in the desired frame `f`.

yuvaltassa, 2024-08-01:
Thanks @MiaoDragon @kevinzakka 

Please feel free to send a PR improving the documentation.

closed 2024-08-01 by yuvaltassa

MiaoDragon, 2024-08-01:
@kevinzakka 
Thanks a lot for the reply! I think the problem is not about the frame where the Jacobian is computed, but that the Jacobian is not the same as the spatial Jacobian defined in the textbook. The position part of the Jacobian follows analytical Jacobian, while the rotation part follows the spatial or maybe a body Jacobian. I understand that the computed one is in the world frame.
Please let me know if my understanding is correct. Thanks!

@yuvaltassa 
I'm not familiar with creating PR for improvement on GitHub, but I can try it out. Thanks!