google-deepmind/mujoco #1868: No collision between bodies

Anubhav0602, 2024-08-01:
Hi,

I'm a student and I'm trying to use MuJoCo for my Research on brachiation.

I'm looking for some help related to coliision/interaction between bodies. 
In given video below you can see insted of hanging on horizontal bar the hook false i.e the horizontal bar act as transparent to hook. what is the possible reason and solution to this problem?

Here is a model which explains my question:
 
 
<mujoco model="acrobot">
  <compiler autolimits="true" meshdir="assets"/>

  <option integrator="implicitfast" gravity="0 0 -9.81"/>

  <include file="./common/visual.xml"/>
  <include file="./common/skybox.xml"/>
  <include file="./common/materials.xml"/>

<asset>
    <material name="white" rgba="1 1 1 1"/>
    <material name="gray" rgba="0.753 0.753 0.753 1"/>
    <material name="black" rgba="0.149 0.149 0.149 1"/>  
    <mesh name="brachiation_hook" file="new_Final_assembly.stl" scale="0.1 0.1 0.1"/>
</asset>

  <default>
    <default class="acrobot">
      
      <geom type="mesh" material="white" contype="1" conaffinity="1"/>
      <joint axis="0 1 0" range="-6.28319 6.28319" frictionloss=".5"/>
      <general biastype="affine" ctrlrange="-6.28319 6.28319"/>
      <default class="size1">
        <joint damping=".01"/>
        <general gainprm="1500" biasprm="0 -1500 -150" forcerange="-50 50"/>
      </default>
    </default>
  </default>

  <default>
    <joint damping=".01"/>
    <geom type="capsule" mass="1"/>
  </default>

  

  <worldbody>
 <light diffuse=".5 .5 .5" pos="-1 0 7" dir="0 0 -1"/>
 <light diffuse=".5 .5 .5" pos="0 0 7" dir="0 0 -1"/>
 <light diffuse=".5 .5 .5" pos="1 0 7" dir="0 0 -1"/>
 <light diffuse=".5 .5 .5" pos="2 0 7" dir="0 0 -1"/>
  <light diffuse=".5 .5 .5" pos="6 0 7" dir="0 0 -1"/>
 <light diffuse=".5 .5 .5" pos="3 0 7" dir="0 0 -1"/>
 <light diffuse=".5 .5 .5" pos="4 0 7" dir="0 0 -1"/>
 <light name="light" pos="0 0 7" dir="0 0 -1"/>

    <body pos="-0.5 0 2.6" euler="90 0 0">
      <joint type="hinge" axis="0 0 1" pos="0 0 0"/>
      <geom type="cylinder" size="0.05 1" rgba="1 0.647 0 1" mass="10"/>
    </body>

    <body pos="0 0 2.6" euler="90 0 0">
        <joint type="hinge" axis="0 0 1" pos="0 0 0"/>
        <geom type="cylinder" size="0.05 1" rgba="1 0.647 0 1" mass="10" contype="1" conaffinity="1"/>
    </body>

    <body pos="0.5 0 2.6" euler="90 0 0">
        <joint type="hinge" axis="0 0 1" pos="0 0 -0"/>
        <geom type="cylinder" size="0.05 1" rgba="1 0.647 0 1" mass="10"/>
    </body>

    <body pos="1 0 2.6" euler="90 0 0">
        <joint type="hinge" axis="0 0 1" pos="0 0 -0"/>
        <geom type="cylinder" size="0.05 1" rgba="1 0.647 0 1" mass="10"/>
      </body>

    <body pos="1.5 0 2.6" euler="90 0 0">
          <joint type="hinge" axis="0 0 1" pos="0 0 -0"/>
          <geom type="cylinder" size="0.05 1" rgba="1 0.647 0 1" mass="10"/>
    </body>

    <body pos="2 0 2.6" euler="90 0 0">
          <joint type="hinge" axis="0 0 1" pos="0 0 -0"/>
          <geom type="cylinder" size="0.05 1" rgba="1 0.647 0 1" mass="10"/>
    </body>

    <body pos="2.5 0 2.6" euler="90 0 0">
          <joint type="hinge" axis="0 0 1" pos="0 0 -0"/>
          <geom type="cylinder" size="0.05 1" rgba="1 0.647 0 1" mass="10"/>
    </body>
 
    <geom name="floor" size="6 3 .2" type="plane" material="grid"/>
    <site name="target" type="sphere" pos="0 0 4" size="0.2" material="target" group="3"/>
    
    <camera name="fixed" pos="0 0 2" zaxis="0 -1 0"/>
    <camera name="lookat" mode="targetbodycom" target="upper_arm" pos="0 0 3"/>

    <body name="hook_one" pos="0.05 0 2.35" euler="0 0 90" childclass="acrobot">
      
      <joint name="hook_one" pos="0 0 2.7" type="free"/>
      <geom mesh="brachiation_hook" mass="0.2" contype="1" conaffinity="1"/>
      
      <body name="upper_arm" pos="0.25 0.05 0" euler="0 180 -90">
        <joint name="shoulder" type="hinge" axis="0 1 0"/>
        <geom name="upper_arm" type="cylinder" fromto="0 0 0 0 0 .5" size="0.05 0.4" material="self" rgba="0 1 0 1" mass="1"/>
      
        <body name="lower_arm" pos="0 0 .5" euler="0 0 0">
          <joint name="elbow" type="hinge" axis="0 -1 0"/>
          <geom name="lower_arm" type="cylinder" fromto="0 0 0 0 0 .5" size="0.05 0.4" material="self" rgba="1 0 0 1" mass="1"/>
        
          <body name="hook_two" pos="0.05 -0.25 0.50" euler="0 0 90" childclass="acrobot">
            
            <joint name="hook_two" type="hinge" axis="0 0 -1" pos="0 0.15 -0.05"/>
            <geom mesh="brachiation_hook" mass="0.2"/>
          </body>
        </body>
      </body>  
    </body>

  </worldbody>

 

	<actuator>
		<motor joint="elbow" name="torque" gear="1" ctrllimited="true" ctrlrange="-5000 5000"/>  
		<position name="position_servo" joint="elbow" kp="5000"/>
		<velocity name="velocity_servo" joint="elbow" kv="10"/>
	</actuator>  

</mujoco>
 

Here is a screenshot / video, illustrating my question:

https://github.com/user-attachments/assets/d535cdfb-ec22-4692-9829-636da3ff82ab

yuvaltassa, 2024-08-01:
Hit 'H' to see the convex hull of you mesh and you'll understand what is happening.

You need to construct the hook from convex pieces, like this https://youtu.be/5k0_wsIRAFc

Anubhav0602, 2024-08-01:
i am unable to understand what does it mean to make from convex pieces, i just want hook to be hanged from horizontal bar. i have designed hook in fusion 360.
here is video after pressing 'H'(Concex hull)

https://github.com/user-attachments/assets/ca117dc1-0205-4375-9ec8-6b0d54f4defc

Anubhav0602, 2024-08-02:
I have modeled this hook in Fusion 360. Do you have any suggestions on how I can make this hook using convex pieces? 
I am attaching a video with the convex hull for your reference.

https://github.com/user-attachments/assets/9ac738aa-5aec-4f26-90fa-8def9fadcc8f