google-deepmind/mujoco #1869: Incorrect acceleration values for pulley system

wykang2, 2024-08-01:
Hi,

I'm a student that is new to MuJoCo and I'm trying to use MuJoCo to simulate a pulley system.

More specifically, I am looking to model problem 34c. from image below:

[image]

The expected values for the acceleration are also given in the textbook as shown:

[image]

I'm looking for some help with my acceleration measurements for my pulley system, which are inconsistent with the values in the solution.

Here is my model for problem 34c:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="test">
  <compiler angle="radian" autolimits="true"/>
  <option>
    <flag sensor="enable"/>
  </option>
    <option gravity="0 0 -9.81"/>
  <visual>
    <headlight diffuse="0.7 0.7 0.7"/>
  </visual>
  <default class="main">
    <joint limited="true" solimplimit="0.95 0.95 0.1 0.5 2" range="0 60"/>
    <geom size="0.01 0 0" rgba="0.9 0.7 0.1 1"/>
    <site rgba="0.9 0.9 0.9 1"/>
  </default>
  <asset>
    <texture type="2d" name="checks1" builtin="checker" rgb1="1 1 1" rgb2="0 0 0" width="256" height="256"/>
    <material name="test_color" texture="checks1" texuniform="true" texrepeat="2 2"/>
  </asset>
  <worldbody>
    <camera name="fixed" pos="0 -1.3 .5" xyaxes="1 0 0 0 1 2"/>

    <!-- left weight -->
    <body name="leftweight" pos="-0.1 0 -0.2" gravcomp="0">
      <joint name="t1" pos="0 0 -0.02" axis="0 0 1" limited="false" type="slide"/>
      <geom size="0.03" pos="0 0 -0.32" mass="1" rgba="0.2 0.2 0.5 1"/>
      <site name="s1" pos="0 0 -0.32"/>
    </body>
    <!-- Pulley 1 (top) -->
    <body pos="0 0 0.3" quat="0.707107 -0.707107 0 0" gravcomp="0">
      <joint name="Pulley" pos="0 0 0" axis="0 0 1" range="0 1.0472"/>
      <geom name="g1" size="0.02 0.015" type="cylinder" material="test_color" rgba="0.3 0.9 0.3 0.4"/>
      <site name="side1" pos="0 -0.03 0" quat="0.707107 0.707107 0 0"/>
    </body>
    <!-- right weight -->
    <body name="tes" pos="0.05 0 0.32" gravcomp="0">
      <joint name="t3" pos="0 0 -0.02" axis="0 0 1" limited="false" type="slide"/>
      <geom size="0.03" pos="0 0 -0.32" mass="2" rgba="0.2 0.2 0.5 1"/>
      <site name="s3" pos="0 0 -0.32"/>
    </body>

    <!-- Pulley 2 (bottom) -->
    <body pos="-0.1 0 -0.3" gravcomp="0" quat="0.707107 -0.707107 0 0">
      <joint name="slider" type="slide" pos="0 0 0" axis="0 1 0" range="-60 60"/>
      <geom name="gg1" size="0.02 0.015" type="cylinder" material="test_color" rgba="0.3 0.9 0.3 0.4"/>
      <site name="sside1" pos="0 -0.03 0" quat="0.707107 0.707107 0 0"/>
      <site name="right" pos="0.02 0 0" quat="0.707107 0.707107 0 0"/>  
      <site name="left" pos="-0.02 0 0" quat="0.707107 0.707107 0 0"/> 
    </body>
    <!-- top left (fixed) -->
    <body pos="-0.2 0 0.3" quat="0.707107 -0.707107 0 0" gravcomp="0">
      <geom name="gg4" size="0.02 0.015" type="cylinder" material="test_color" rgba="0.3 0.9 0.3 0.4"/>
      <site name="sside4" pos="0 -0.03 0" quat="0.707107 0.707107 0 0"/>
    </body>

  </worldbody>
  <equality>
    <tendon tendon1="tendon1" polycoef="0 1 0 0 0"/>
    <tendon tendon1="tendon3" polycoef="0 1 0 0 0"/>
    <!-- <tendon tendon1="tendon2" polycoef="0 1 0 0 0"/> -->
  </equality>
  <tendon>
    <spatial name="tendon1" damping="100" frictionloss="100" width="0.002" rgba="0.95 0.3 0.3 1">
      <!-- <site site="s1"/> -->
      <site site="sside4"/>
      <site site="left"/>
      <site site="right"/>
      <geom geom="g1" sidesite="side1"/>
      <site site="s3"/>
      
    </spatial>
    <spatial name="tendon3" damping="100" frictionloss="100" width="0.002" rgba="0.95 0.3 0.3 1">
      <!-- <site site="s1"/> -->
      <site site="s1"/>
      <site site="sside1"/>
    </spatial>
  </tendon>
  <actuator>
    <position name="slider" joint="slider" ctrlrange="-.05 .05" kp="100"/>
  </actuator>
  <sensor>
    <tendonpos tendon="tendon1" name="rope_length"/>
    <tendonpos tendon="tendon3" name="rope_length2"/>
    <accelerometer name="accel_left" site="s1"/>
    <accelerometer name="accel_tes" site = "s3"/>
  </sensor>
</mujoco>
```

</details>

Here is a video, illustrating what the model looks like when ran:

https://github.com/user-attachments/assets/8ebb7e58-0298-442c-82d3-3e948472657d

In the model, I used equality constraints on the tendons to keep their lengths constant. 

To model the acceleration of each mass, I used two accelerometers and plotted out the accelerations over time. 

Here is the acceleration over time graph for the right mass (initially moving downwards):

[image]

Here is the acceleration over time graph for the left mass (initially moving upwards):

[image]

I mainly care about the accelerations of the pulley from the start of the simulation to when the right mass reaches its low point and the left mass reaches its high point (where a real pulley system would stop moving). This corresponds to around the 0 second to 1 second sections of the graphs above. 

My issue is that the not only do the magnitudes of the values seem incorrect accordingly to the answer to the problem above, the accelerations keep changing as well. 

Any help with solving or identifying possible issues would be greatly appreciated.

Balint-H, 2024-08-07:
I can only provide a partial answer:
The angle of the tendon will affect the accelerations, so you should align them so they are vertical. Also, no hinge joint needed in the pulley object itself. And its better to model the bottom pulley/object B as a rigid object instead of two separate connected by a pulley. Lastly, the accelerometer sensors are biased by the gravity, you need to subtract 9.81 from the vertical component (depending on the accelerometer orientation of course).

What I can't give an answer for is the oscillatory behaviour, which does not purely arise from "bouncing off" the limits. Could someone help clear this up?

[image]
Why is the above configuration in equilibrium? The ratio of the equilibrium height of the bodies seem to depend on the mass ratio and happens with both equality constraints and tendon length ranges.

mihirp1998, 2024-08-09:
great thank you so much! 

Do u know if there is a easy way to visualize these constraint violation forces that get generated, so that it can help debug in the future more easily?

Balint-H, 2024-08-09:
`data->qfrc_actuator` holds the force actuators put on degrees of freedom. Since you used a position actuator, it will always apply a non-zero proportional force on your system when it deviates from the specified position. This is not a hidden effect of the actuator, this is what its used for! You could change its gearing or gains to zero to disable it temporarily.

Constraint forces are stored in
`data->qfrc_constraint` and `data->efc_force` (in joint and constraint space respectively)

mihirp1998, 2024-08-12:
great thanks :) !

wykang2, 2024-08-13:
@Balint-H Thank you so much for the help! The new system you created seemed to be working on similar systems. However, the limitation I am currently facing is due to the mass being attached to the pulley for the bottom pulley. I am currently trying to create a similar system, but with another pulley attached to the bottom pulley instead of a mass. 

Here is an image of the pulley system I am talking about:
[image]

Below is some minimal xml for the pulley system on the leftside of the image above based on the code you provided in your original reply:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="test">
  <compiler angle="radian" autolimits="true"/>
  <option>
    <flag sensor="enable"/>
  </option>
    <option gravity="0 0 -9.81"/>
  <visual>
    <headlight diffuse="0.7 0.7 0.7"/>
  </visual>
  <default class="main">
    <joint limited="false"solimplimit="0.95 0.95 0.1 0.5 2" range="-0.4 0.3"/>
    <geom size="0.01 0 0" rgba="0.9 0.7 0.1 1"/>
    <site rgba="0.9 0.9 0.9 1"/>
  </default>
  <asset>
    <texture type="2d" name="checks1" builtin="checker" rgb1="1 1 1" rgb2="0 0 0" width="256" height="256"/>
    <material name="test_color" texture="checks1" texuniform="true" texrepeat="2 2"/>
  </asset>
  <worldbody>
    <camera name="fixed" pos="0 -1.3 .5" xyaxes="1 0 0 0 1 2"/>

    <!-- left weight -->
    
    <!-- Pulley 1 (top) -->
    <body pos="-0.06 0 0.3" quat="0.707107 -0.707107 0 0" gravcomp="0">
      <geom name="   " size="0.02 0.015" type="cylinder" material="test_color" rgba="0.3 0.9 0.3 0.4"/>
	  <site name="sside3" pos="0 -0.03 0" quat="0.707107 0.707107 0 0"/>
      <site name="right2" pos="0.02 0 0" quat="0.707107 0.707107 0 0"/>  
      <site name="left2" pos="-0.02 0 0" quat="0.707107 0.707107 0 0"/> 
    </body>
    <!-- right weight -->
    <body name="tes" pos="-0.04 0 0 " gravcomp="0">
      <joint name="t3" axis="0 0 1" limited="false" type="slide"/>
      <geom name="m=2" size="0.0252" rgba="0.2 0.2 0.5 1" mass="2"/>
      <site name="s3"/>
    </body>

    <!-- Pulley 2 (bottom) -->
    <body pos="-0.1 0 -0.1" gravcomp="0" quat="0.707107 -0.707107 0 0">
      <joint name="slider" type="slide" pos="0 0 0" axis="0 1 0"/>
      <geom name="  " size="0.02 0.015" type="cylinder" material="test_color" rgba="0.3 0.9 0.3 0.4" mass="0.1"/>
      <site name="sside2" pos="0 0.03 0" quat="0.707107 0.707107 0 0"/>
      <site name="right" pos="0.02 0 0" quat="0.707107 0.707107 0 0"/>  
      <site name="left" pos="-0.02 0 0" quat="0.707107 0.707107 0 0"/> 
	  <!-- <body name="leftweight" pos="0 0.1 0" gravcomp="0">
        <geom name ="m=1" size="0.02" pos="0 0 0" rgba="0.2 0.2 0.5 1" mass="1"/>
        <site name="s1" pos="0 0 0" euler="1.5708 0 0"/>	
	  </body> -->
    </body>
    <!-- top left (fixed) -->
    <body pos="-0.15 0 0.3" quat="0.707107 -0.707107 0 0" gravcomp="0">
      <geom name=" " size="0.02 0.015" type="cylinder" material="test_color" rgba="0.3 0.9 0.3 0.4" density="10000"/>
      <site name="sside4" pos="0.03 0 0" quat="0.707107 0.707107 0 0"/>
    </body>
 
    <!-- new top left pulley -->
    <body pos="-0.3 0 0.3" quat="0.707107 -0.707107 0 0" gravcomp="0">
      <geom name="    " size="0.02 0.015" type="cylinder" material="test_color" rgba="0.3 0.9 0.3 0.4" density="10000"/>
      <site name="new" pos="0.03 0 0" quat="0.707107 0.707107 0 0"/>
    </body>

    <!-- new bottom pulley -->
    <body pos="-0.22 0 -0.4" gravcomp="0" quat="0.707107 -0.707107 0 0">
      <joint name="slider2" type="slide" pos="0 0 0" axis="0 1 0"/>
      <geom name="     " size="0.02 0.015" type="cylinder" material="test_color" rgba="0.3 0.9 0.3 0.4" mass="0.1"/>
      <site name="new2" pos="0 0.03 0" quat="0.707107 0.707107 0 0"/>
      <site name="right4" pos="0.02 0 0" quat="0.707107 0.707107 0 0"/>  
      <site name="left4" pos="-0.02 0 0" quat="0.707107 0.707107 0 0"/> 
    </body>

  </worldbody>
  
  <equality>
    <tendon tendon1="tendon1" polycoef="0 1 0 0 0"/> 
    <tendon tendon1="tendon2" polycoef="0 1 0 0 0"/> 
  </equality>

  <tendon>
    <spatial name="tendon1" width="0.002" rgba="0.95 0.3 0.3 1" range="0 1">
      <site site="sside4"/>
	  <geom geom="  " sidesite="sside2"/> 
      <site site="left2"/>
	  <geom geom="   " sidesite="sside3"/>
      <site site="right2"/>
      <site site="s3"/>
      
    </spatial>

    <!-- Tendon for new pulley -->
    <spatial name="tendon2" width="0.002" rgba="0.95 0.3 0.3 1" range="0 1">
      <site site="new"/>
	  <geom geom="     " sidesite="new2"/> 
      <site site="sside2"/>
    </spatial>

  </tendon>
  <sensor>
    <!-- <accelerometer name="accel_left" site="s1"/>
    <accelerometer name="accel_tes" site = "s3"/> -->
    <force name="f1" site="s3"/>
  </sensor>
</mujoco>
```

</details>

The main issue is that in order to create this type of system, the pulley mass has to be non-zero since we removed the mass attached to the bottom pulley. However, doing so (possibly along with other factors), will cause the values to be incorrect. Are there any workarounds or solutions for this? In other words, is there any way to create a massless bottom pulley without directly attaching the weight to it so that our system can be extended to my example attached?

Thank you so much.

Balint-H, 2024-08-13:
As far as the first system you show in your figure, attaching the left mass to the centre of P1 via a rope or another rigid connection  makes no difference, it is only a matter of visual notation for the illustration. Therefore you can treat pulley P1 and the left mass as a single object and you don't have a zero inertia problem.

With your second system shown I agree it is a bit trickier. I'd start with setting the P2 geom's density to a tiny amount, increasing it with an order of magnitude if it is unstable. (Alternatively you can set the density to zero and add an inertial component.

mihirp1998, 2024-08-19:
Thanks @Balint-H for all the help!

> EDIT:
> 
> Apologies for missing something obvious... The actuator you added is always active, and acts as a spring in your system, which is one of the causes for oscillations. The others are the angled tendons. In terms of the offset in your readings, keep in mind that accelerometers include the gravity as an offset (like real accelerometers). After a short initial transient period (a couple of frames), the reworked model below gives exactly 1/3 g and 2/3 g accelerations!
> 
> Reworked xml:
> Original answer:

With your reworked xml we get the following plots, we get accurate values for acceleration and tension after some point in time, as you pointed out - 1/3 g and 2/3 g. However There is an initial bump (marked in the red box) for the first few milliseconds  in Tension and accelerations. 

I was wondering what causes this bump, is it becoz the tension force only gets activated when there is a change in length in the tendon which happens after few milliseconds?

[image]

Balint-H, 2024-08-19:
Hello,

Yes, that transient period is because of the softness of constraints in the engine (essentially as if the cable had some elasticity). You can explore the effects of manipulating the `solref` and `solimp` parameters which governs the impedance of the constraints (see: https://mujoco.readthedocs.io/en/latest/modeling.html#solver-parameters)! You can make it harder if you need a shorter transient period.

mihirp1998, 2024-08-20:
i see thanks!

closed 2024-08-30 by kevinzakka