google-deepmind/mujoco #1870: Recompile is not doing in-place update of model and data

bikcrum, 2024-08-01:
Branch: main (9febd760)

I am adding a geom to the existing model and running`recompile(model, data)` to obtain the updated model. However, the updates are not being applied in-place contrary to the documentation https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-recompile. This issue is also mentioned in https://github.com/google-deepmind/mujoco/issues/1822, which was closed without any comment. 

### Code to reproduce:

```
import mujoco as mj

# Define a simple model
model_string = """
    <mujoco>
        <worldbody>
            <body>
                <freejoint/>
                <geom size=".15" mass="1" type="sphere"/>
            </body>
        </worldbody>
    </mujoco>
"""

# Compile the model
spec = mj.MjSpec()
spec.from_string(model_string)
model = spec.compile()
data = mj.MjData(model)

# Add a geom to the world body
body = spec.worldbody
geom = body.add_geom()
geom.name = 'my_geom'
geom.size[0] = 1
geom.pos = [1, 1, 0]

# Recompile the model
new_model, new_data = spec.recompile(model, data)

new_model.geom('my_geom')  # No error

model.geom('my_geom')
```

### Error:
```
# KeyError: "Invalid name 'my_geom'. Valid names: []"
```

quagla, 2024-08-01:
It happens in place only in C (which is the function described in the docs), not in Python.

quagla, 2024-08-01:
Here is how the C function is wrapper in the bindings if you wish to have a look at what happens under the hood 

https://github.com/google-deepmind/mujoco/blob/main/python/mujoco/structs.cc#L454

bikcrum, 2024-08-01:
@quagla Is it possible to edit this function to perform in-place operations for Python as well? Is there something preventing this?

quagla, 2024-08-07:
They will not be true in-place operations anyway, causing possible issues with variables referencing the old mjData which would become invalid without the user explicitly being aware of it.