google-deepmind/mujoco #1873: I have modeled this hook in Fusion 360. Do you have any suggestions on how I can make this hook using convex pieces?

Anubhav0602, 2024-08-02:
I have modeled this hook in Fusion 360. Do you have any suggestions on how I can make this hook using convex pieces? 
I am attaching a video with the convex hull for your reference.

https://github.com/user-attachments/assets/9ac738aa-5aec-4f26-90fa-8def9fadcc8f

_Originally posted by @Anubhav0602 in https://github.com/google-deepmind/mujoco/issues/1868#issuecomment-2264550930_

Anubhav0602, 2024-08-02:
i want my hook to hang from the above horizontal bar

v-r-a, 2024-08-02:
https://github.com/SarahWeiii/CoACD

Balint-H, 2024-08-06:
I'd also like to mention the option of using SDFs (https://github.com/google-deepmind/mujoco/tree/main/plugin/sdf)

And also disabling collisions for your mesh, and manually or procedurally fitting invisible collision primitives. You can use [replicate](https://github.com/google-deepmind/mujoco/tree/881b777ffd5e95aa36867f93b0f6b27921da53d5/model/replicate) If your hook part has a region of constant curvature. 

If you need something quick, and for only this model, then the collision primitive method may be the fastest. (And may give you better contacts then with a decomposed mesh!)

yuvaltassa, 2024-08-09:
Also https://github.com/kevinzakka/obj2mjcf