google-deepmind/mujoco #1875: mj_ray performance regression from 3.1.1 to 3.2.0

michael-projectx, 2024-08-02:
I've recently upgraded from v3.1.1 to v3.2.0 and I'm seeing some large regressions in performance in simulations that call `mj_ray`. We use `mj_ray` to determine the visibility of certain objects in a scene. This post is twofold:
1. Perhaps we're using mj_ray inefficiently and there are some optimizations required on our end.
2. Perhaps there's been a regression in `mj_ray`.

For all of our simulations we track simulated time and wall clock time and provide a "realtime ratio" (wall clock time / simulated time) as an output. I noticed that many of our simulations showed little difference in realtime ratio between both mujoco version, but some were quite different. Using one of the simulations that showed a large regression in wall clock time (the sim takes much longer to run), I did some comparisons.

 I've run the same simulation using 3.1.1 and 3.2.0 to compare performance. Here is some data I've collected:

For both simulations the simulated time is 12.102 seconds
When running with v3.1.1, the wall clock time is XXX seconds (a ratio of X.XX)
When running with v3.2.0, the wall clock time is 110.86 seconds (a ratio of 9.16)

In order to dig deeper, I've run the above simulation using both versions of mujoco under `perf` to try to pinpoint what was running slower. It turns out that `mj_ray` is the culprit.

When running with v3.1.1, the calls to `mj_ray` account for 13.9% of the total execution time of the simulation
When running with v3.2.0, the calls to `mj_ray` account for 78.6% of the total execution time of the simulation

Diving deeper, `mju_rayTree` accounts for 13.1% of the simulation time for v3.1.1 and 78.3% of the simulation time for v3.2.0

`ray_triangle` which is called by `mju_rayTree` takes less that 1% under both versions.

What additional information can I provide in order to help debug this issue?

quagla, 2024-08-05:
Hi, could you provide the value of `model.nbvh` for your model before and after the change?

michael-projectx, 2024-08-05:
With mujoco 3.2.0 we have:
  model.nbody=96
  model.nbvh=2622528
  model.nbvhstatic=2622528
  model.nbvhdynamic=0

With mujoco 3.1.1 we have:
  model.nbody=96
  model.nbvh=2622599
  model.nbvhstatic=2622599
  model.nbvhdynamic=0

So, pretty similar, but not exactly he same.

quagla, 2024-08-07:
How much time is going into `mju_raySlab` for the two cases? Or where otherwise is the time going?

michael-projectx, 2024-08-07:
Let me dig back up the perf data, but from the screenshot I have, I see this:
[image]
I don't actually see a call to mju_raySlab, so maybe it's very tiny?

We have a function to check if a point is visible (creatively named `isPointVisible` to check if a point in our scene is visible from a camera location. We use `mj_ray` for this (perhaps there is a better way. Here's the timing breakdown that I can find (I'm going to invert the call stack from what is shown in the image):

`isPointVisible` : 75.3%
`mj_ray` : 75.3%
`mj_rayMesh` : 75.1%  | `ray_box` : 0.0501%
`mju_rayTree` : 75% |  `ray_box` : 0.0631%
`??` : 0.0958% | `ray_triangle` : 0.0636

Items on the same line appear to both be called by the function on the line above it. There's a bunch of stuff that seems to be called by the `??` function above, but it and all of the called functions are optimized out. I can try to rebuild with debug symbols and see if I get more data. I'm not sure if any of that helps. What function should be calling `mju_raySlab`?

michael-projectx, 2024-08-07:
Here's another view of all function calls that match `_ray`:
[image]

quagla, 2024-08-08:
Could you please set this flag https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-global-bvactive to `false` and repeat the benchmark? That's the only line of code that has changed in mj_rayMesh since 3.1.1.

nimrod-gileadi, 2024-08-08:
Could you also provide `np.max(model.bvh_depth)` (or the equivalent from C)?

If possible, it would help if you gave us an example model where this happens, and the code you're using for measurement.

michael-projectx, 2024-08-09:
I'm using [`perf`](https://en.wikipedia.org/wiki/Perf_(Linux)) to measure the performance. The only difference between my A/B comparison between 3.1.1 and 3.2.0 is the mujoco version. We use bazel for our build system, so I'm simply fetching a different tarball from the github repository.

`max(model.bvh_depth)` is 18 for mujoco 3.2.0 and 19 for 3.1.1

I'll try to put together a reproducible model that I can share.

quagla, 2024-08-09:
> Could you please set this flag https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-global-bvactive to `false` and repeat the benchmark? That's the only line of code that has changed in mj_rayMesh since 3.1.1.

Did this make any difference at all?

michael-projectx, 2024-08-09:
> > Could you please set this flag https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-global-bvactive to `false` and repeat the benchmark? That's the only line of code that has changed in mj_rayMesh since 3.1.1.
> 
> Did this make any difference at all?

I missed your original message. I just gave it a try.

`bvactive = "true"`
simulation time = 12.10200, wall time = 113.72063, ratio: 9.39685

`bvactive = "false"`
simulation time = 12.10200, wall time = 110.67355, ratio: 9.14506

Looks like the difference is negligible (probably within the noise of system resource contention).

michael-projectx, 2024-08-16:
When would we expect a new release to be created containing #1906?

yuvaltassa, 2024-08-17:
Look at our historical release cadence on the changelog. Early September. In the meantime build from source.