google-deepmind/mujoco #1877: [Python MuJoCo 3.2] Load Failed when load 100 or more materials in xml

forworksmk, 2024-08-04:
Using MuJoCo version 3.1.6 in python, When I load my XML file, no error occurs.

However, after updating MuJoCo to version 3.2 and loading an XML file, I encounter the following error:

`FatalError: Maximum number of materials is 100`

What can I do to avoid this error or maybe solve this error?

yuvaltassa, 2024-08-04:
What is the XML file (and why do you have so many materials)?

forworksmk, 2024-08-04:
Hello,

I have built a simulation environment for indoor spaces using an XML file, and it worked well with MuJoCo version 3.1.6. However, upon upgrading to version 3.2, I encountered an issue where the simulator indicates that the maximum number of materials has been exceeded.

In my environment, there are approximately 200 materials. My question is: Is there a way to exceed the max number of materials limit in MuJoCo 3.2, or are there any recommended approaches to handle this limitation?

Thank you for your assistance.

closed 2024-08-06 by copybara-service[bot]

nimrod-gileadi, 2024-08-06:
We'll increase the limit back to 1000 in the upcoming 3.2.1.

nimrod-gileadi, 2024-08-07:
Unfortunately the commit just missed the cut for 3.2.1 and will be pushed in 3.2.2.

nimrod-gileadi, 2024-08-08:
3.2.2 is now released and the issue should be fixed.

commit 9db9df73 (2024-08-06): Increase material count limit to 1000.

The texture limit used to be 1000, but when materials with multiple textures were introduced in 3.2.0, the limit was reduced to 100. Since there are model out there that already have more than 100 and fewer than 1000 textures/materials, we should maintain backwards compat.

Fixes #1877

PiperOrigin-RevId: 659901629
Change-Id: If0f6d0500d1d808bc3ec2d87d4a3faf24007da59