google-deepmind/mujoco #1879: MJX action space size setting

ChanMook, 2024-08-05:
Hi,

I'm a engineer and I'm working on MuJoCo MJX.

I'm looking for some help with action space size setting.

i want to learn for the parameters of control algorithm.

but, now action space set only number of acturator.

how can i change the size of action apace?

btaba, 2024-08-05:
Hi, to change the size of the action space, consider adding or removing actuators from your model

closed 2024-08-05 by btaba

ChanMook, 2024-08-05:
Thanks your comment.
However, I would like to know how to make the size of the actuator and action different.

btaba, 2024-08-05:
By size of the actuator, I'm assuming you mean `mjModel.nu` "number of actuators/controls = dim(ctrl)"?
See https://mujoco.readthedocs.io/en/stable/XMLreference.html#actuator for the XML reference