google-deepmind/mujoco #1885: Replicate

MaabSadig, 2024-08-08:
I am trying to build a robot from 4 identical modules by using Replicate.
Right now the modules are on top of each other as needed but when moving one module the ones above it do not move. How to link the frame of the end of each module with the frame of the beginning of the subsequent one?

quagla, 2024-08-08:
If I understand the question, the answer is that the frame needs to be inside a body of a module in order to be attached to it.

closed 2024-08-13 by quagla

MaabSadig, 2024-08-16:
What I want is to connect the end of each module with its subsequent one. As I mentioned before, now with replicate the modules are not connected they are just on top of each other but they move independently to each other.