google-deepmind/mujoco #1888: XML error: multiple orientation specifiers are not allowed

Louis2099, 2024-08-09:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/google-deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a student and I'm trying to use MuJoCo for kinova gen3 robot arm simulation.

"ValueError: XML Error: multiple orientation specifiers are not allowed
Element 'inertial', line 43" 
occurred when I was loading the following XML file with:
 "self.model = mujoco.MjModel.from_xml_path(base_path)".
After looking into mujoco documentation I believe orientation specifiers are "quat, axisangle, xyaxes, zaxis, euler", and within the XML file only quat is used, thus I'm confused about the following error.

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="JACO3_XML_V11">
	<compiler angle="radian"/>
	<size njmax="500" nconmax="100"/>
	<asset>
		<!-- Gen3 Arm Meshes -->
		<mesh name="Base_Link" file="gen3/Base_Link.stl"/>
		<mesh name="Shoulder_Link" file="gen3/Shoulder_Link.stl"/>
		<mesh name="HalfArm1_Link" file="gen3/HalfArm1_Link.stl"/>
		<mesh name="HalfArm2_Link" file="gen3/HalfArm2_Link.stl"/>
		<mesh name="ForeArm_Link" file="gen3/ForeArm_Link.stl"/>
		<mesh name="SphericalWrist1_Link" file="gen3/SphericalWrist1_Link.stl"/>
		<mesh name="SphericalWrist2_Link" file="gen3/SphericalWrist2_Link.stl"/>
		<mesh name="Bracelet_Link" file="gen3/Bracelet_Link.stl"/>
		<!-- Robotiq Hand Meshes -->
		<mesh name="robotiq_85_base_link_coarse" file="robotiq/robotiq_85_base_link_coarse.stl"/>
		<mesh name="outer_knuckle_coarse" file="robotiq/outer_knuckle_coarse.stl"/>
		<mesh name="outer_finger_coarse" file="robotiq/outer_finger_coarse.stl"/>
		<mesh name="inner_knuckle_coarse" file="robotiq/inner_knuckle_coarse.stl"/>
		<mesh name="inner_finger_coarse" file="robotiq/inner_finger_coarse.stl"/>
		<material name="BeigeMaterial" specular="0" shininess="0" reflectance="0" rgba="0.26 0.16 0.16796875 1" emission="0" />
	</asset>

    <actuator>
        <motor name="" joint="Actuator1" gear="1 0 0 0 0 0" ctrlrange="0 0.8727" forcerange="-39 39"/>
        <motor name="HalfArm1_Link_motor" joint="Actuator2" gear="1 0 0 0 0 0" ctrlrange="0 0.8727" forcerange="-39 39"/>
        <motor name="HalfArm2_Link_motor" joint="Actuator3" gear="1 0 0 0 0 0" ctrlrange="0 0.8727" forcerange="-39 39"/>
        <motor name="ForeArm_Link_motor" joint="Actuator4" gear="1 0 0 0 0 0" ctrlrange="0 0.8727" forcerange="-39 39"/>
        <motor name="SphericalWrist1_Link_motor" joint="Actuator5" gear="1 0 0 0 0 0" ctrlrange="0 0.8727" forcerange="-9 9"/>
        <motor name="SphericalWrist2_Link_motor" joint="Actuator6" gear="1 0 0 0 0 0" ctrlrange="0 0.8727" forcerange="-9 9"/>
        <motor name="Bracelet_Link_motor" joint="Actuator7" gear="1 0 0 0 0 0" ctrlrange="0 0.8727" forcerange="-9 9"/>
    </actuator>

	<worldbody>
 		<geom name="floor" pos="0 0 -0.66" size="10 10 1" type="plane" rgba="1 0.83 0.61 1" material="BeigeMaterial" />
        <geom name="wall_1" pos="0 -2.5 1" size="2.5 2.5 1" type="plane" quat="0.707107 -0.707107 0 0" rgba="0.95 0.95 0.95 1" material="BeigeMaterial" />
        <geom name="wall_2" pos="-2.5 0 1" size="2.5 2.5 1" type="plane" quat="0.707107 0 0.707107 0" rgba="0.95 0.95 0.95 1" material="BeigeMaterial" />	
        	
        <body name="Base_Link" pos="0 0 0">
			<inertial pos="-0.000648 -0.000166 0.084487" mass="1.697353" fullinertia="0.004622 0.004495 0.002079 -9E-06 -6E-05 -9E-06"/>
			<geom type="mesh" rgba="0.79216 0.81961 0.93333 1" mesh="Base_Link"/>

		    <body name="Shoulder_Link" pos="6.1995e-05 0 0.15643" quat="-3.67321e-06 -1 0 0">
			**<inertial pos="-2.3e-05 -0.010364 -0.07336" quat="0.707051 0.0451246 -0.0453544 0.704263" mass="1.37735" fullinertia="0.00457 0.004831 0.001409 -1E-06 -2E-06 -0.000448"/>**
			<joint name="Actuator1" pos="0 0 0" axis="0 0 1"/>
			<geom type="mesh" rgba="0.75294 0.75294 0.75294 1" mesh="Shoulder_Link"/>

			<body name="HalfArm1_Link" pos="0 0.005375 -0.12838" quat="0.707105 0.707s108 0 0">
				<inertial pos="-4.4e-05 -0.09958 -0.013278" quat="0.482348 0.516286 -0.516862 0.483366" mass="1.16367" diaginertia="0.0113017 0.011088 0.00102532"/>
				<joint name="Actuator2" pos="0 0 0" axis="0 0 1" limited="true" range="-2.2 2.2"/>
				<geom type="mesh" rgba="0.75294 0.75294 0.75294 1" mesh="HalfArm1_Link"/>
[... 83 lines cut ...]
		<light directional="false" cutoff="90" exponent="1" diffuse="1 1 1" specular="0 0 0" pos="1 0 1.5" dir="-1 0 -2.5"/>
	</worldbody>
	<sensor>
		<jointpos joint="Actuator1" name="jointpos_joint_1"/>
        <jointpos joint="Actuator2" name="jointpos_joint_2"/>
        <jointpos joint="Actuator3" name="jointpos_joint_3"/>
        <jointpos joint="Actuator4" name="jointpos_joint_4"/>
        <jointpos joint="Actuator5" name="jointpos_joint_5"/>
        <jointpos joint="Actuator6" name="jointpos_joint_6"/>
        
        <!-- <jointvel joint="Actuator1" name="jointvel_joint_1"/>
        <jointvel joint="Actuator2" name="jointvel_joint_2"/>
        <jointvel joint="Actuator3" name="jointvel_joint_3"/>
        <jointvel joint="Actuator4" name="jointvel_joint_4"/>
        <jointvel joint="Actuator5" name="jointvel_joint_5"/>
        <jointvel joint="Actuator6" name="jointvel_joint_6"/> -->

		<velocimeter site="Bracelet_Link" name="velocimeter_Bracelet_Link"/>
	</sensor>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

quagla, 2024-08-09:
Apologies, I will improve the error message since it's misleading. It's not possible to use `fullinertia` together with any of the orientation specifiers.

kevinzakka, 2024-08-09:
@Louis2099 FYI, we have a [kinova gen3](https://github.com/google-deepmind/mujoco_menagerie/tree/main/kinova_gen3) in Menagerie.

commit da15aa6a (2024-08-09): Improve error message for fullinertia and orientation specifiers.

Fixes #1888.

PiperOrigin-RevId: 661359190
Change-Id: Ibe38e88024b4a43922352b42347abf1026f168a2
closed 2024-08-09 by copybara-service[bot]