google-deepmind/mujoco #1891: MJX: pd control with target velocity in actuator/position

Rian-Jo, 2024-08-10:
**How to ask for help**

Hi,

I'm looking for some help with MJX actuator/position for the pd control.  As far as i know, mujoco uses qfrc_actuator = kp*(q_d - q) - kd*(qdot) . I just want to add target velocity qdot_d to form  qfrc_actuator = kp*(q_d - q) - kd*(qdot_d - qdot). I've confirmed that a PID plugin exists in MuJoCo, but it seems that MJX does not. 

Is there any method to simply add target velocity or to implement pid plugin? and if user can add actuator plugin, other low level control may be implemented.

kevinzakka, 2024-08-10:
What's stopping you from just using `motor` actuators and setting the torque directly using the PD control law you defined above?

Rian-Jo, 2024-08-12:
@kevinzakka hi, 

how can you handel n_frame? in my knowledge, if mujoco opt.timestep is 1ms and mjx env timestep 5ms, which means that pipelinestep(mujoco step) is run 5 times per 1 mjx environment step. if one uses the motor actuator and write the pd control in mjx step function, pd control is updated in 5ms but i want it is run in 1ms with 5ms of policy updates. do you have any menthod to resolve this?

thank you.

yuvaltassa, 2024-08-12:
If you put two actuators on the same joint their outputs are summed. So make seperate P (position) and D (velocity) actuators, done.

closed 2024-08-12 by yuvaltassa

Rian-Jo, 2024-08-20:
@yuvaltassa  thank you!!

@kevinzakka hi, firstly, i tried to use motor auctuator for pd control but the training result was too bad to keep working on. have you tried to train the agent with motor in 1khz envrionment timestep? 

thank you!!