google-deepmind/mujoco #1893: Question about getting the site speed in mujoco

Yingfan99327, 2024-08-12:
Hi,

I'm a student and I'm trying to use MuJoCo for robotic simulation.
I'm looking for some help with how to get the site speed in mujoco.

I found the function "mj_objectVelocity“ to get the site speed, here are my questions:

1) Is this the right function to get the speed of the site I defined on some rigid bodies, are there any methods to get the site speed in mujoco?
2) According to the explanation of the function that "Compute object 6D velocity (rot:lin) in object-centered frame, world/local orientation." Does it mean that the site velocity vector obtained from this function is expressed in the site frame not the world frame? Or the rot speed is expressed in the site frame while the linear speed is expressed in the global frame?

Thanks for your time and wishing you all the best.

nimrod-gileadi, 2024-08-12:
1. This is the right function to use, although you could use a `framelinvel` sensor to do this, which might be easier.
2. Velocities are expressed in a frame centered on the site, but the orientation of that frame depends on `flg_local`.

Yingfan99327, 2024-08-12:
Ah, I see, thanks so much for your reply.

Yingfan99327, 2024-08-12:
Thanks for your reply.

So, it means that both the linear and rotational velocities are expressed in the frame centered on the site I define, while the orientation of the site frame can be expressed w.r.t the global frame or the local (site) frame depends how I set the flg_local.

Actually, what I am trying to do is to read the xvel and xacc of the flange site I attached to the end-effector of a robot by using this mj_objectVelocity function, my expected steps are as follows,

1) Get the flange xvel and xacc with mj_objectVelocity with flg_local = false, which I think the results I obtained represent the velocities and accelerations of the flange expressed in the global frame;
2) Calculating the flange xvel and xacc with the rotation matrix and replacement vector between the global frame and the robot base frame.

Is this right?

I guess if I set the flg_local = true, and I hope to get the flange xvel and xacc expressed in the global frame, I need to do some calculation with the site_mat of the flange site and then to do some calculations, so setting flg_local = false is much easier which I can directly get the flange xvel and xacc expressed in the global frame, right?

Really appreciate your reply and wishing you all the best.