google-deepmind/mujoco #1896: Resetting equality constraints with spec.recompile()

connie-liou, 2024-08-13:
Hello! I'm using MuJoCo to model a sort of robotic deployment process/assembly line, which requires me to reset the position of an actuator and change equality constraints as each piece is pushed out from a slider. Basically, I want to attach a "pusher" to a part, push it out and release it, then attach the next piece to the same pusher. 

I know there's issues with spec.recompile(), but I'm hoping to use the python bindings to make changes to the equality constraints on the fly. 

Currently running into pointer errors when I try to call a specific equality constraint and set it to inactive, then recompile it. 

```
    deployed_part_equality = spec.equality[0]
    deployed_part_equality.active = 0

    model = spec.recompile(model,data)
    data = mj.MjData(model)
```

I immediately get ```pointer being freed was not allocated```. Maybe there's a smarter way to frame the problem that doesn't require recompiling, but any help would be appreciated!

quagla, 2024-08-13:
Hi see here for how to use recompile https://github.com/google-deepmind/mujoco/blob/main/python/mujoco/specs_test.py#L175

Let me know if you still get the issue if you do `model_new, data_new = spec.recompile(model_old, data_old)`.

connie-liou, 2024-08-13:
> Hi see here for how to use recompile https://github.com/google-deepmind/mujoco/blob/main/python/mujoco/specs_test.py#L175
> 
> Let me know if you still get the issue if you do `model_new, data_new = spec.recompile(model_old, data_old)`.

Still getting the same issue :/

yuvaltassa, 2024-08-13:
You don't need to do this. Equality constraints can already be enabled/disabled on the fly using `mjData.eq_active`. mjSpec not required.

closed 2024-08-13 by yuvaltassa

connie-liou, 2024-08-14:
For generating the equality constraints using spec, how is the anchor point specified?

yuvaltassa, 2024-08-14:
This has been discussed before in several other issues (at least one that I remember). The relevant field is eq_data, search for that.

connie-liou, 2024-08-15:
I think I'm still misunderstanding how relpose and anchors are used. I have an object that I want to move (a hinge on a slider), and then fix the hinge at a specific point in space with the weld equality constraint as described in the XML reference here, then detach the hinge from the slider. This is a closed kinematic loop, so I have to use equality constraints to connect the last hinge child to the slider :
"Name of the second body. If this attribute is omitted, the second body is the world body. Welding a body to the world and changing the corresponding component of mjData.eq_active at runtime can be used to fix the body temporarily."

Every time I activate the constraint during runtime, the body snaps back to it's initial position instead of its current position. I've tried to set the anchor to the desired position, but it still snaps back to the initial position. I've also tried creating a dummy body and joint at the desired position in space and attempting to weld/connect/joint equality to that, but the best I could get is the hinge circles the desired point at a radius of its initial distance even after playing with the first 3 entries of relpose. 

I think I'm missing something, but any help would be very useful!

yuvaltassa, 2024-08-16:
Maybe the code [here](https://github.com/google-deepmind/mujoco/blob/cfc7dc984897e7e9efe912720672ae0f67f94dcf/src/xml/xml_native_reader.cc#L1903) will help clarify?

xkiixkii, 2024-09-13:
> Maybe the code [here](https://github.com/google-deepmind/mujoco/blob/cfc7dc984897e7e9efe912720672ae0f67f94dcf/src/xml/xml_native_reader.cc#L1903) will help clarify?

Hi, I don't know if my understanding of eq_data is correct. This code means that eq_data[:3] is anchor, eq_data[3:10] is relpos, and eq_data[10] is torquescale. Is this right?

I have also noticed that in this [discussion](https://github.com/google-deepmind/mujoco/discussions/1208#discussioncomment-7607311) you have mentioned that eq_data will be documented [here](https://mujoco.readthedocs.io/en/stable/computation/index.html#user-inputs) and the documentation is still not updated.

yuvaltassa, 2024-09-13:
1. Yes you are correct.
2. That was a mistype, I meant that `eq_active` should be documented, which it is.
3. @thowell is planning to provide a new parametrization for weld constraints, like the one we just added for connect constraints (Item 5 in the 3.2.3 [changelog](https://mujoco.readthedocs.io/en/latest/changelog.html#version-3-2-3-sep-12-2024)), which will not use `eq_data` at all (except for `torquescale`), but rather align the frames of two sites. This will make modifying welds much easier. We will update this issue when that happens.

yuvaltassa, 2024-09-16:
Both connect and weld constraints can now be defined using sites, this will be out in the next release (or build from source)

commit 60a1921b (2024-09-16): Add site semantics to weld constraint. Fixes #1896

PiperOrigin-RevId: 675112981
Change-Id: Ie4c01440d54116075df89f480591700762892cbd