google-deepmind/mujoco #1897: Creating a Massless Movable Pulley

mihirp1998, 2024-08-13:
Hi,

I'm a student and I'm trying to use MuJoCo for research.

I'm looking for some help with creating a massless movable pulley, specifically the pulley on the left in the figure below. When i reduce the mass of the pulley to mujoco's minimum value, the  system breaks as i think the acceleration of the pulley becomes very high. If i add mass to the pulley it won't be accurate wrt the system i wish to simulate:

[image]

A possible solution would be to get rid of the tendon and directly attach the movable pulley body with the left sphere, so that i dont have to define a massless pulley. However this wont workout for more complex pulley systems like these:

[image]

Any form of help or ideas would be very helpful.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="test">
  <compiler angle="radian" autolimits="true"/>
  <option>
    <flag sensor="enable"/>
  </option>
    <option gravity="0 0 -9.81"/>
  <visual>
    <headlight diffuse="0.7 0.7 0.7"/>
  </visual>
  <default class="main">
    <joint limited="false" solimplimit="0.95 0.95 0.1 0.5 2" range="-0.4 0.3"/>
    <geom size="0.01 0 0" rgba="0.9 0.7 0.1 1"/>
    <site rgba="0.9 0.9 0.9 1"/>
  </default>
  <asset>
    <texture type="2d" name="checks1" builtin="checker" rgb1="1 1 1" rgb2="0 0 0" width="256" height="256"/>
    <material name="test_color" texture="checks1" texuniform="true" texrepeat="2 2"/>
  </asset>
  <worldbody>
    <camera name="fixed" pos="0 -1.3 .5" xyaxes="1 0 0 0 1 2"/>

    
    <!-- Pulley 1 (top) -->
    <body pos="-0.06 0 0.3" quat="0.707107 -0.707107 0 0" gravcomp="0">
      <geom name="   " size="0.02 0.015" type="cylinder" material="test_color" rgba="0.3 0.9 0.3 0.4"/>
	  <site name="sside3" pos="0 -0.03 0" quat="0.707107 0.707107 0 0"/>
      <site name="right2" pos="0.02 0 0" quat="0.707107 0.707107 0 0"/>  
      <site name="left2" pos="-0.02 0 0" quat="0.707107 0.707107 0 0"/> 
    </body>
    <!-- right weight -->
    <body name="tes" pos="-0.04 0 0 " gravcomp="0">
      <joint name="t3" axis="0 0 1" limited="false" type="slide"/>
      <geom name="m=2" size="0.0252" rgba="0.2 0.2 0.5 1" mass="2"/>
      <site name="s3"/>
    </body>

    <!-- Pulley 2 (bottom) -->
    <body name="massless" pos="-0.1 0 -0.4" gravcomp="0" quat="0.707107 -0.707107 0 0">
      <joint name="slider" type="slide" pos="0 0 0" axis="0 1 0"/>
      <geom name="  " size="0.02 0.015" type="cylinder" material="test_color" rgba="0.3 0.9 0.3 0.4" mass="0.01"/>
      <site name="sside2" pos="0 0.03 0" quat="0.707107 0.707107 0 0"/>
      <site name="sside5" pos="0 0.03 0" quat="0.707107 0.707107 0 0"/>
      <site name="right" pos="0.02 0 0" quat="0.707107 0.707107 0 0"/>  
      <site name="left" pos="-0.02 0 0" quat="0.707107 0.707107 0 0"/> 

    </body>
    <!-- top left (fixed) -->
    <body pos="-0.15 0 0.3" quat="0.707107 -0.707107 0 0" gravcomp="0">
      <geom name=" " size="0.02 0.015" type="cylinder" material="test_color" rgba="0.3 0.9 0.3 0.4" density="10000"/>
      <site name="sside4" pos="0.03 0 0" quat="0.707107 0.707107 0 0"/>
    </body>

    <!-- left weight -->
    <body name="leftweight" pos="-0.1 0 -0.45" gravcomp="0">
     <geom name ="m=1" size="0.02" pos="0 0 0" rgba="0.2 0.2 0.5 1" mass="1"/>
     <site name="s1" pos="0 0 0" euler="1.5708 0 0"/>	
     <joint name="leftslide" axis="0 0 1" limited="false" type="slide"/>
    </body> 

  </worldbody>
  
  <equality>
    <tendon tendon1="tendon1" polycoef="0 1 0 0 0"/> 
    <tendon tendon1="tendon2" polycoef="0 1 0 0 0"/> 
  </equality>

  <tendon>
    <spatial name="tendon1"  width="0.002" rgba="0.95 0.3 0.3 1">
      <site site="sside4"/>
	  <geom geom="  " sidesite="sside2"/> 
      <site site="left2"/>
	  <geom geom="   " sidesite="sside3"/>
      <site site="right2"/>
      <site site="s3"/>
    </spatial>
    <spatial name="tendon2"  width="0.002" rgba="0.95 0.3 0.3 1">
      <site site="sside5"/>
      <site site="s1"/>
    </spatial>    
  </tendon>
  <sensor>
    <force name="f1" site="s3"/>
    <tendonpos tendon="tendon1" name="tendon1_length"/>
    <accelerometer name="acceleration_left" site="s1"/>
    <accelerometer name="acceleration_right" site = "s3"/>    
  </sensor>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

https://github.com/user-attachments/assets/f69e9bc2-1cfd-4505-b8a5-70ddb7a316a2

yuvaltassa, 2024-08-13:
Sorry, massless moving bodies are not supported (and as you say, not realistic).

closed 2024-08-13 by yuvaltassa

Balint-H, 2024-08-15:
A note on system 5-E12: Since the floating pulley's state is wholly determined by the height of m1, you could model the scene by removing the fixed pulley, placing m1 above the floating pulley and setting them as a single rigid body, then reversing gravity (e.g. with the gravcomp attribute) for m1.

mihirp1998, 2024-08-18:
i see, that makes sense.. thanks!