google-deepmind/mujoco #1908: Misleading MJX documentation

wladekpal, 2024-08-18:
Hi, 

I've been recently working with MJX model where I had Cylinders colliding with Boxes. In the [feature parity documentation](https://mujoco.readthedocs.io/en/latest/mjx.html#feature-parity) it is specified that "`ELLIPSOID` and `CYLINDER` are implemented but only collide with other primitives.". 
Since my understanding is that Box is a primitive I thought that it should work, but I got an error that such a collision type is not supported. 
I later saw that there is a note in a table with features under development about this collision type, but I think that it is very misleading to list it under "fully supported" and it would probably be a good idea to make it more precise.

I also wanted to ask is there any ETA for this feature ?

yuvaltassa, 2024-08-18:
Yes, boxes are basically implemented as meshes in MJX, this is indeed a documentation bug.

We are currently working on a large overhaul of collisions in both MuJoCo and MJX, which should include this feature.

wladekpal, 2024-08-18:
Ok, thanks for a quick response.

commit 8272211d (2024-08-30): Update MJX documentation for ellipsoid and cylinder collision support.

fixes #1908

PiperOrigin-RevId: 669244021
Change-Id: I5f757562be333b5b9923655d155d51ed3dbd09c4
closed 2024-08-30 by copybara-service[bot]