google-deepmind/mujoco #1911: A small question about how to calculate joint torque to decrease joint velocities and accelerations more quickly?

Yingfan99327, 2024-08-19:
Hi,

I'm a student and I'm trying to use MuJoCo for robotic simulation.

I'm looking for some help with the joint torque command.

I used the motor-type actuators in panda robot simulation. Here is my problem.

```
# joint velocities and joint accelerations read from data.qvel and data.qacc at time t
qvel =  [ 3.249e-02  5.866e-01  2.5216e-02 -2.1146e+00 -5.746e-02 -1.286e+00 -1.1046e-04] 
qacc =  [ 6.3916e-02  5.0266e+00 -9.9666e-02 -8.999e+00 1.756e-01  3.289e+00  8.253e-04] 
# joint torque computed from impedance controller at time t
data.ctrl = [-3.259e+03 -5.652e+04 -2.531e+03  2.154e+05 5.769e+03  1.280e+05  6.300e+00]

Then after mj_step, time_step = 0.001

# joint velocities and joint accelerations read from data.qvel and data.qacc at time t+1
qvel = [ 3.255e-02  5.917e-01  2.511e-02 -2.1237e+00 -5.731e-02 -1.280e+00 -1.097e-04] 
qacc = [ 6.361e-02  5.066e+00 -1.017e-01 -8.971e+00 1.783e-01  3.368e+00  8.287e-04] 
```
It could be seen that the joint acceleration decreases due to the joint torque with right signs,but what I expect is that the joint acceleration could be decreased much more quickly so that the joint velocities could reach 0 more quickly.

And, yes, the joint torque seems very high due to the high damping coefficients I set for the robot. However, I set the ctrl range in the xml file, I guess after I set the ctrl range, the actual torque sent to the joint will be automatically clipped into this range, right? Is this the reason why the joint acceleration changes not too much after mj_step?

```
<actuator>
      <motor class="panda" name="panda_joint1" joint="joint1" ctrlrange="-87 87" gear="1"/>
      <motor class="panda" name="panda_joint2" joint="joint2" ctrlrange="-87 87" gear="1"/>
      <motor class="panda" name="panda_joint3" joint="joint3" ctrlrange="-87 87" gear="1"/>
      <motor class="panda" name="panda_joint4" joint="joint4" ctrlrange="-87 87" gear="1"/>
      <motor class="panda" name="panda_joint5" joint="joint5" ctrlrange="-12 12" gear="1"/>
      <motor class="panda" name="panda_joint6" joint="joint6" ctrlrange="-12 12" gear="1"/>
      <motor class="panda" name="panda_joint7" joint="joint7" ctrlrange="-12 12" gear="1"/>
</actuator>
```
So, how could I enable the joint velocities and accelerations decrease to 0 quickly with the right joint torque?

Thanks and wishing you all the best.

yuvaltassa, 2024-08-19:
Did you try removing the `ctrlrange`?

Yingfan99327, 2024-08-19:
And as I read from this issue #1095, can I use the method mentioned in this issue,
```
data.joint("my_joint").qfrc_constraint + data.joint("my_joint").qfrc_smooth
```
to measure the real joint torque applied to each joint after I individually set the joint torque equal to the results from the joint impedance controller? Namely,
```
data.ctrl = tau_from_joint_impedance_controller
mj_step(model, data)
print(data.joint("my_joint").qfrc_constraint + data.joint("my_joint").qfrc_smooth)
```
Because I suspect, because the joint torque computed is so large, with the ctrl range limits set in each actuator, the actual joint torques applied to each actuator is actually very small, which result in the very slow decreasing speed of joint accelerations and joint velocties.

Yingfan99327, 2024-08-19:
Hi, Yuval, 

Thanks for your reply. I removed the ctrlrange defined in the actuators, but it presented the same result.

yuvaltassa, 2024-08-19:
Why are you using your own controller I suspect your actuators are inheriting the `ctrlrange` from [here](https://github.com/google-deepmind/mujoco_menagerie/blob/39664247032283fc5cda18cf6ce2b28cdc84eb61/franka_emika_panda/panda.xml#L10)

Yingfan99327, 2024-08-19:
Yes, I noticed it so I commented out this line as I showed in the xml file above, but it also didn't work.

yuvaltassa, 2024-08-19:
In that case I'm not sure...

Anyway, why are you implementing you own actuator rather than using the built-in ones?

Yingfan99327, 2024-08-19:
Well, I just hope to make the simulation close to the real scenes, I mean, hope that the joint position limits and joint torque limits listed in the official web as https://frankaemika.github.io/docs/control_parameters.html could be well meet during the simulation. 

You mean the actuators defined originally in the mujoco_menagerie? I tried just now, it doesn't work neither. And it seems lack some torque limits on joint torques.

Thanks for your time and help, Yuval. May I just make a small summary of some questions I am confused here so that you can give me a quick check?
1) The only thing that will affect the input torque would be the ctrlrange, right? Without the ctrl range, the joint torque would be the torque you send to the actuator, otherwise it will be clamped to the control range.
2) In real scenarios, I guess friction also plays an important role, so would there be something I can modify or add about joint friction to help me plan the joint motion as I expect?
3) If I need to use the self.data.qfrc_inverse.copy()[0:self.joint_num] to calculate the gravity compensation torque to help the robot maintain the configuratuon, I need to make sure the joint velocities and joint accelerations are all zeros.
4) If everything works fine, it definitely possible to realize a classic joint impedance controller or admittance controller in mujoco, I mean, to realize some expected tasks such as 
a) Follow a defined trajectory in cartesian space and stop at the end position.
b) Move when some external force is applied to the end-effector and stop quickly and smoothly when the external force is removed.
Because I have already realized some tasks in simulation, however, because I use this simulation tool not for a long time, I am a bit worried whether I miss some important settings in my model files or other aspects. 

Btw, I also referred to some other developers' codes in panda robot simulation, however, most of them simply use methods like mocap, position-type actuators instead of motor-type actuators because we have different focus in simulation.

Anyway, thanks very much for your time and help.

Wishing you all the best.