google-deepmind/mujoco #1917: How to use the 'cable' composite object for modelling crane and payload system

jessicatodd, 2024-08-21:
Hi Mujoco community!

I'm very new to Mujoco and am working on a project design control systems for a crane system. We want to use the cable composite to model our crane's cable down to its payload, since the cable has nonneglible mass and we will be operating the crane in a fairly dynamic environment (dynamics of the cable itself will come into effect). I'm trying to get a simple model of the crane, with a payload and cable attached up and running but am struggling with the cable composite plugin. 

I don't fully understand how to enforce attachment points between the end of the cable and the crane, and the end of the cable and the payload, so would appreciate some assistance. 

My very simple xml is below and the run script attached. 

[cable_demo.zip](https://github.com/user-attachments/files/16697983/cable_demo.zip)

Mordween, 2024-08-22:
Hi,

I use this to attach the body to the cable:

```html

<body name="payload" pos="0 0.015 0.02">
      <geom type="sphere" size=".001 .001" density="100" rgba=".5 .2 .8 1" mass="0.01"/>
      <freejoint/>
    </body>

<composite type="cable" count="50 1 1" curve="s 0 0" offset="0 0 0" initial="ball" size="0.39 1 1">
    <joint kind="main" damping="0.01" armature="0.001"/>
    <geom type="capsule" size=".0005 .001 0.3" rgba=".8 .2 .1 1"/>
</composite>

<equality>
    <weld body1="payload" body2="B_last" active="true" anchor="0 0 0" torquescale="0"/>
</equality>
```

Your cable is defined with many small bodies, and if you want to refer to a specific one, you can use B_0 for the first segment, B_N for the Nth segment, or B_last for the last segment.

I hope this helps

jessicatodd, 2024-08-22:
If you look the code I attached, I do use the equality constraint with B_last. The issue is that there is a gap between the position of the payload and the end of the cable, and if I lengthen the cable, it does not seem to reduce this gap (essentially the payload is suspended in free space below the end of the cable"

quagla, 2024-08-23:
If I understand it correctly, it's because constraints are soft in mujoco. You can use `solref` and `solimp` in the constraint parameters to make it stiffer.

Balint-H, 2024-08-29:
It is simpler than that, the connect constraint assumes it is satisfied in the starting position. Move the payload to the end of the cable in your xml, and it will get rid of the offset! (e.g. pos="0.2 0 0.2")