google-deepmind/mujoco #1921: How to select points on the flexcomp-grid fixed on the geometry？

walt666, 2024-08-23:
Hi,

I'm a student and I'm trying to use MuJoCo for Robot simulation.

I'm looking for some help with XML.

Here is a model which explains my question:
I want to use flexcomp-grid to build the strips and then attach the ends of the strips to the two boxes, but I can't select the ends of the grid to connect to the boxes. flexcomp can only be fixed in space with a pin check point. I tried to pin the strip to the box, but as soon as the box moved, the strip separated from the box.

<details>
  <summary>minimal XML</summary>

<mujoco>
  <include file="scene.xml"/>
  <extension>
    <plugin plugin="mujoco.elasticity.shell"/>
    <plugin plugin="mujoco.elasticity.membrane"/>
  </extension>

  <option timestep="0.001" solver="CG" tolerance="1e-6" integrator="implicitfast"/>

  <size memory="10M"/>

  <visual>
    <map stiffness="100"/>
  </visual>

  <default>
    <default class="wall">
      <geom type="plane" size=".5 .5 .05"/>
    </default>
  </default>

  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1"/>
    <body name="box1" pos="0 0 0.3">
      <joint name="pull1" type="slide" axis="0 1 0" damping=".01"/>
      <geom name="box11" type="box" size=".3 .3 .3" rgba="0 .9 0 1"  />
    </body>  

    <body name="CB" pos="-1 0 0.6">
      <flexcomp type="grid" count="42 2 1" spacing="0.035 0.035 0.035"  
              radius=".001" mass="10" name="plate" dim="2" rgba=".8 .2 .1 1">
        <pin id="0 1 82 83" />
        <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001"/>
        <edge equality="false" damping="10"/>
        
        <plugin plugin="mujoco.elasticity.membrane">
          <config key="poisson" value="0.2"/>
          <config key="thickness" value="1e-2"/>
        
          <config key="young" value="3e6"/>
        </plugin>
      </flexcomp>
    </body>

    <body name="box3" pos="-1 2 0.3">
      <joint name="pull3" type="slide" axis="0 1 0" damping=".01"/>
      <geom type="box" size=".3 .2 .3" rgba="0 .9 0 1" />
    </body>
   
    <body name="box2" pos="-2 0 0.3">
      <joint name="pull2" type="slide" axis="0 1 0" damping=".01"/>
      <geom type="box" size=".3 .3 .3" rgba="0.9 0 0 1" />
    </body>
  </worldbody>
  
  <actuator>
    <adhesion name="adhere_cam" body="box1" ctrlrange="0 1" gain="10"/>
    <general name="pullgo" joint="pull1" ctrlrange="-100 100" ctrllimited="true"/>
  </actuator>

</mujoco>

</details>

Here is a screenshot / video, illustrating my question:
[image]
[image]

quagla, 2024-08-23:
Hi, since pinning just removes degrees of freedom, you can only pin to a parent, not to two independent bodies. You can either add two equality constraints or pin at one end and use an equality constraint at the other one.

walt666, 2024-08-23:
> Hi, since pinning just removes degrees of freedom, you can only pin to a parent, not to two independent bodies. You can either add two equality constraints or pin at one end (by making the flex a child of one of the two boxes) and use an equality constraint at the other one.

Thank you. I already tried that. But the equality constraint requires selecting the body, and I can't select a point on the other side of the grid. When the equality constraint selects the body of grid, but that's not the point I want.

quagla, 2024-08-27:
You can try loading and saving the model, in this way you will see that `flexcomp` is expanded into the bodies that are forming it and find out the name of the body that you want to use for the constraint.