google-deepmind/mujoco #1929: build error from source

LeiHanyue, 2024-08-27:
Hi,

I'm a engineer and I'm trying to use MuJoCo for simulation.

I'm looking for some help with build.

my environment: 
    ubuntu22.04, 
    gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
    g++ (Ubuntu 7.5.0-6ubuntu2) 7.5.0
    clang version 10.0.0-4ubuntu1
    cmake version 3.22.1
    
I made some changes to the source code because the [-Werror=unused-variable] error, which is showed in the first picture. And then I encountered the following issue(s).

[  0%] Built target lodepng
[  1%] Built target ccd
[  6%] Built target qhullstatic_r
[  7%] Built target tinyxml2
[  7%] Built target tinyobjloader
[  9%] Linking CXX shared library lib/libmujoco.so
lto1: internal compiler error: compressed stream: data error
unrecognized DWARF version in .debug_info at 6
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
lto-wrapper: fatal error: /usr/bin/c++ returned 1 exit status
compilation terminated.
/usr/bin/ld.gold: fatal error: lto-wrapper failed
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/mujoco.dir/build.make:1095: lib/libmujoco.so.3.2.2] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1605: CMakeFiles/mujoco.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

So I want to know is there anything wrong with my environment? What I should do to solve this error?
[image]

LeiHanyue, 2024-08-27:
Maybe there's something wrong with the picture, but the changes are just adding lines like 'static_cast<void>(name); // Silence -Wunused-variable.' Nothing else has been modified.

yuvaltassa, 2024-08-28:
Did you try deleting [this line](https://github.com/google-deepmind/mujoco/blob/72a0ab145d30e9241e08569d690b39f79db9a1e8/cmake/MujocoOptions.cmake#L93)?

LeiHanyue, 2024-08-29:
I found that all the errors disappeared after I upgraded g++ from 7.5.0 to 11， even after I deleted “-Wunused” in https://github.com/google-deepmind/mujoco/blob/72a0ab145d30e9241e08569d690b39f79db9a1e8/cmake/MujocoOptions.cmake#L93. But I didn't delete that line before.

closed 2024-09-04 by yuvaltassa