google-deepmind/mujoco #1930: Wrong HField vertical scaling in unity plugin

kezzyhko, 2024-08-27:
# Description

_Video with the problem is attached below_

When using a terrain (hfield geom) in Unity, physics collisions may not match the rendered visuals. This happens in case when terrain data contains heights in smaller range than [0..1]. For example, if terrain is a flat plane at height 0.7, mujoco will think that highest point is at height 1, and collision will happen at height 0.7 * 0.7 = 0.49. But as soon as we add one point at height 1, everything starts working correctly.

# Steps to reproduce

1. Create empty project in Unity and add MuJoCo package
2. Import the attached .unitypackage file ([HFieldBug.zip](https://github.com/user-attachments/files/16767225/HFieldBug.zip))
3. Open the scene and hit play (notice how the cube falls through the terrain, but then stops)
4. Stop simulation and tick the "Add Workaround Point" on "TerrainCreator" object
5. Hit play (notice that the cube no longer falls through)

# Video

https://github.com/user-attachments/assets/e3e6b9f9-d32d-403b-b1a3-ddc5871701d5

# Versions

_It probably does not depend on particular versions, but I'm gonna include them anyway_

**OS**: Windows 10
**MuJoCo**: 3.2.0
**Unity Editor**: 2022.3.28f1

Balint-H, 2024-08-28:
Hello!

Thanks for the top quality bug report! I implemented this feature originally with normalized heightmaps in mind which is why I originally missed this issue. I am aware of this bug, and I think this commit fixes it (and add inspector tools to add terrain automatically): https://github.com/Balint-H/mujoco/tree/feature/unity-hfield-context

There's an open PR for it that has not yet been merged.

However I've mainly tested it in my own use cases and I'd really appreciate if you could give it a go as well (the commit is for MuJoCo 3.1.4). If it still doesn't work I'll go through the scenes you shared and send in a fix ASAP.

If it does work I'll ask for someone at DM to merge in the PR.

I'm glad others are using the plugin as well, so let me know if you have other questions.

kezzyhko, 2024-08-28:
@Balint-H 
Yep, I have seen your PR, but for some reason I didn't think it would fix that bug. Anyway, I have tested my usecase with MuJoCo 3.1.4 + plugin at `https://github.com/Balint-H/mujoco.git?path=unity#feature/unity-hfield-context`, and it works perfectly. Now waiting for the PR merge :) Thanks for your help and for your contributions!

kezzyhko, 2024-08-28:
#1504

SebasNorenaR, 2026-10-23:
fixed in https://github.com/google-deepmind/mujoco/pull/1504

closed 2026-10-23 by SebasNorenaR