google-deepmind/mujoco #2035: MJX implementation of mj_jacSubtreeCom

gautica, 2024-08-31:
### The feature, motivation and pitch

I am working on implementing a whole-body-controller for bipedal robots. It is important to have the task for the angular and linear momentum for the Center-of-Mass. In mujoco, I can get such information from **mj_jacSubtreeCom** and **mj_angmomMat**. I would like to ask if it is possible to implement the two functions in MJX, or please give me some hints how to implement them. Thank you very much.

btaba, 2024-09-09:
Hi @gautica , can you use [`subtreelinvel `](https://mujoco.readthedocs.io/en/stable/XMLreference.html#sensor-subtreelinvel) and `subtreeangmom` sensors [recently](https://github.com/google-deepmind/mujoco/commit/20296466b102cfd898cc2f7fda2fd923ffcd5434) added to MJX? Let us know if that's sufficient for your use-case

Rian-Jo, 2024-10-17:
@btaba hi, can you help regarding subtreelinvel? tried to use it but it is not going well.
can you please check https://github.com/google-deepmind/mujoco/discussions/2147#discussion-7329000

thank you.

btaba, 2024-10-24:
Closing the issue, and replied on the discussion

closed 2024-10-24 by btaba