google-deepmind/mujoco #2039: Error: engine error: mj_copyData: attempting to copy mjData while stack is in use

KingBright, 2024-09-04:
### Intro

Hi!

I am just a programmer, I use MuJoCo for .

### My setup

Mujoco version: 3.2.3
env: Ubuntu 22.04 on a Virtual Machine on Apple Sillicon
API: C++

### What's happening? What did you expect?

I just want to copy the mjData in a plugin's compute callback. Following is what I did
```
void Test::Compute(const mjModel *m, mjData *d, int instance)
{
    mjData data_copy = mj_makeData(m);
    mj_copyData(data_copy, m, d);
}
```

And it reports the following error
`
engine error: mj_copyData: attempting to copy mjData while stack is in use
`

I wonder what is it mean, and is there a right timing to do the copy.

### Steps for reproduction

1. Write a new plugin
2. In the compute funciton, call the mj_dataCopy function to copy the mjData
3. Run the simulate gui and just load the plugin
4. You will see the error msg

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-09-04:
Ya, you can't do that inside a plugin because... the stack is in use :slightly_smiling_face: 

Let's start with the basics, **why** do you think you want to copy mjData?

I'm closing this because this WAI, you should try to copy mjData when the stack is in use. Here's [some context regarding the internal stack](https://mujoco.readthedocs.io/en/latest/programming/simulation.html#internal-stack)

closed 2024-09-04 by yuvaltassa

KingBright, 2024-10-17:
Forgot to reply this.

I just wanted to do some parallel simulation with the data in mjData without modify the original one, so I tried to copy the mjData.
Finally I found that I could make a new mjData from mjModel and read the data from the original one.

yuvaltassa, 2024-10-17:
more generally, use the functions we have for saving and reloading [the state](https://mujoco.readthedocs.io/en/latest/computation/index.html#the-state).

MuJoCo has a very well-defined state. Use it! 🙂

fangqi-Zhu, 2024-11-21:
When I use `get_state` and `set_state`, I noticed that there is floating-point precision error in the state. I’d like to ask if this is normal or a bug.

Here is my implementation. I found that the pixel-level data is consistent, but when comparing the state, I observed slight discrepancies on the order of 1e-14.

```
  def dump_state(self):
      state = np.empty(mujoco.mj_stateSize(self.model, mujoco.mjtState.mjSTATE_INTEGRATION))
      mujoco.mj_getState(self.model, self.data, state, spec=mujoco.mjtState.mjSTATE_INTEGRATION)
      return state
  
  def load_state(self, state):
      mujoco.mj_setState(self.model, self.data, state, spec=mujoco.mjtState.mjSTATE_INTEGRATION)
      mujoco.mj_forward(self.model, self.data
```