google-deepmind/mujoco #2044: Mujoco Python to Unity

mg638, 2024-09-06:
### The feature, motivation and pitch

hello. I would like to demonstrate the process of reproducing mujoco training data in Python and reproducing that data into a Unity Mujoco model using Unity's Mujoco plugin. Is there anything that can be helpful when creating a feature like this? Even if you try to find the information, it's not easy.

Balint-H, 2024-09-06:
Hello,

Can you explain a bit more what exactly are you trying to do? Transfer a control policy learned in python to Unity? Record motion trajectories as data in python, and use them/replay them in Unity?

mg638, 2024-09-09:
Thank you for your reply.
To be exact, the Mujoco model is learned with Pytorch and a PT file is created. After that, I plan to use the learned PT file to play motion in Unity's Mujoco model.