google-deepmind/mujoco #2049: ASAN build failure from mj_markStack/mj_freeStack

michael-ahn, 2024-09-09:
### Intro

I am using mujoco within a larger bazel project and an unable to compile mujoco from source with asan checks enabled. Wondering if the mujoco team has run into this before if they have a similar setup internally.

### My setup

- MuJoCo 3.2.2, C/C++
- clang/clang++ 14.0.0
- Ubuntu 20.04 LTS

### What's happening? What did you expect?

When `ADDRESS_SANITIZER` is defined, `mj_markStack` and `mj_freeStack` (in `engine_io.h`) are instead defined as `mj__markStack` and `mj__freeStack`: [engine_io.h#L126](https://github.com/google-deepmind/mujoco/blob/4862b9e765daf2881686c806bf37a5a7930e3001/src/engine/engine_io.h#L126)
- The single underscore versions are then defined in `mujoco.h`: [mujoco.h#L1783](https://github.com/google-deepmind/mujoco/blob/4862b9e765daf2881686c806bf37a5a7930e3001/include/mujoco/mujoco.h#L1783)
- Most of the existing callers of `mj_{mark/free}Stack` within mujoco do not include `mujoco.h`, so they do not see the asan-specific definition and will cause a `implicit-function-declaration` compiler warning/error.

I have a workaround fix here [[5cb397fe50]](https://github.com/michael-ahn/mujoco/commit/5cb397fe501ddd75031e9ca28b7c50334ff9f931), where I simply include `mujoco.h` where `mj_{mark/free}Stack` is used so the asan-specific definition is used correctly, and can confirm this works as expected.
- It doesn't feel like `mujoco.h` is intended to be widely included internally like this so I'm not a big fan of this fix; happy to defer to the mujoco team on what this should look like.

### Steps for reproduction

1. Compile mujoco with asan flags. Minimal flags to trigger this error on mujoco's cmake setup is `-DADDRESS_SANITIZER -fsanitize=address`.

This produces many errors like the following:
```
mujoco/src/engine/engine_collision_driver.c:659:3: error: implicit declaration of function 'mj_markStack' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
  mj_markStack(d);
```

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-09-09:
Thanks @michael-ahn,
 
TBH it was never clear to me why these ASAN redefinitions were in `mujoco.h` to begin with... 
@nimrod-gileadi, can you see any reason not to inline these into engine_io.h?

nimrod-gileadi, 2024-09-10:
I think the reason to put them in mujoco.h is so that code outside the engine can use the mj_markStack API and get the same ASAN checks.

We don't run into this issue internally is that we include mujoco.h as Michael suggests, but we strip it out when we export to GitHub. In the past this was fine, and ensured that dependencies within the engine code were explicitly listed, but after b04de39e4e07f7a988aabd390cad31806a5ec153 this `include` is required for ASAN.

I think the solution is to move the declarations and ASAN definition of the stack functions into a new header file, mjstack.h, which is included from `mujoco.h`, and include that file from engine code that uses the stack.

commit e4d41533 (2024-09-10): Move sanitizer instrumentation to a separate header file. Fixes #2049.

PiperOrigin-RevId: 672986547
Change-Id: I42522f4925237a73168e965364a2dd65c0f067cb
closed 2024-09-10 by copybara-service[bot]