google-deepmind/mujoco #2053: mujoco.FatalError: gladLoadGL error

akajiejie, 2024-09-12:
I am a student and I am training my machine learning model on a Linux server. I have encountered some problems that I cannot solve.
The environment configuration I use is as follows：
[image]
Is this a problem with the mujoco version or other issues?

KyberRobot, 2024-09-25:
I had some problems with running a server as well. I am using Ubuntu 24.04. After updating the mesa package, I am able to render now using mesa and egl. 

The steps I did are the following:

```
sudo add-apt-repository ppa:kisak/kisak-mesa
sudo apt update
sudo apt-get install libegl-mesa0
```

On the server there is no X-server running, so I use xvfb (install via `sudo apt-get install xvfb`). You can test if OpenGL is rendering using this command then (you will need to install `mesa-utils` via apt-get for `glxgears`):

```
xvfb-run -a glxgears # Test if working
```

The python code runs then via

```
xvfb-run -a python ...
```

where my python code looks something like this:

```python
%env MUJOCO_GL=egl

import matplotlib.pylab as plt

import mujoco
from mujoco import viewer

model = mujoco.MjModel.from_xml_path('scene.xml')
data = mujoco.MjData(model)

renderer = mujoco.Renderer(model)

mujoco.mj_step(model, data)
renderer.update_scene(data)
plt.imshow(renderer.render())
```

Hope this helps.

akajiejie, 2024-10-07:
Thanks to KyberLabsAIRobot for the answer

akajiejie, 2024-10-07:
Thanks to KyberLabsAIRobot for the answer
I found in subsequent development that this might be a problem caused by the computing server not having a remote desktop
and I solved this problem
The steps I did are the following:
apt update && apt install -y libglu1-mesa-dev mesa-utils xterm xauth x11-xkb-utils xfonts-base xkb-data libxtst6 libxv1
export TURBOVNC_VERSION=2.2.5
export LIBJPEG_VERSION=2.0.90
wget http://aivc.ks3-cn-beijing.ksyun.com/packages/libjpeg-turbo/libjpeg-turbo-official_${LIBJPEG_VERSION}_amd64.deb
wget http://aivc.ks3-cn-beijing.ksyun.com/packages/turbovnc/turbovnc_${TURBOVNC_VERSION}_amd64.deb
dpkg -i libjpeg-turbo-official_${LIBJPEG_VERSION}_amd64.deb
dpkg -i turbovnc_${TURBOVNC_VERSION}_amd64.deb
rm -rf *.deb

rm -rf /tmp/.X1*  
USER=root /opt/TurboVNC/bin/vncserver :1 -desktop X -auth /root/.Xauthority -geometry 1920x1080 -depth 24 -rfbwait 120000 -rfbauth /root/.vnc/passwd -fp /usr/share/fonts/X11/misc/,/usr/share/fonts -rfbport 6006

then use my local device to connect to the server's remote desktop via SSH
If you encounter problems, please refer to the TurboVNC connection tutorial

closed 2024-10-07 by akajiejie