google-deepmind/mujoco #2061: Clearing `extension/plugin` fields in `mjSpec`

hartikainen, 2024-09-17:
### The feature, motivation and pitch

I'm trying to use the new model editing API to convert plugin actuators into motors but am hitting an error due to dangling  [`extension/plugin`](https://github.com/google-deepmind/mujoco_menagerie/blob/e9a9cf4db26f49d896cdbb846da4070b1e2c32c4/shadow_dexee/shadow_dexee.xml#L56-L93) elements in the spec. It seems like the `mjSpec` doesn't expose the `extension/plugin` fields to me so that I could clear them. Currently, I have to manually modify the raw xml string, and it would be cleaner if I could do this using the model editing API.

### Alternatives

I've tried deleting the `plugin` elements with `plugin_actuator.plugin.delete()`. This segfaults. (See the comment in the reproducing scripts below.)

I've tried looking for `extension` or `plugin`-like elements from the [mjSpec](https://mujoco.readthedocs.io/en/stable/APIreference/APItypes.html#mjspec) but can't seem to find anything, apart from `actuator.plugin` mentioned above, that would allow me to clear those elements.

I can work around this by replacing the `actuator`s and removing the `extension` elements in the raw xml. This works fine but it would be nice to be able to do such replacements in with the model editing API.

### Additional context

Here's a small example of the error:
```python
import mujoco
from absl import app
from robot_descriptions import shadow_dexee_mj_description

def main(unused_argv):
    spec = mujoco.MjSpec()
    spec.from_file(shadow_dexee_mj_description.MJCF_PATH)

    for plugin_actuator in spec.actuators:
        name = plugin_actuator.name
        target = plugin_actuator.target
        forcerange = plugin_actuator.forcerange
        # Need to explicitly instantiate `mjtTrn` because the delete below clears the type.
        trntype = mujoco.mjtTrn(plugin_actuator.trntype)

        # plugin_actuator.plugin.delete()  # This one segfaults
        plugin_actuator.delete()

        motor_actuator = spec.add_actuator()
        motor_actuator.name = name
        motor_actuator.target = target
        motor_actuator.trntype = trntype
        motor_actuator.ctrlrange = forcerange
        motor_actuator.gear[0] = 1.0
        motor_actuator.forcerange = forcerange

    mj_model = spec.compile()  # this one fails.

if __name__ == "__main__":
    app.run(main)
```

This one fails with:
```
python ./mj_spec_remove_actuator.py
Traceback (most recent call last):
  File "mj_spec_remove_actuator.py", line 50, in <module>
    app.run(main)
  File "conda/envs/mj_spec_test/lib/python3.11/site-packages/absl/app.py", line 308, in run
    _run_main(main, args)
  File "conda/envs/mj_spec_test/lib/python3.11/site-packages/absl/app.py", line 254, in _run_main
    sys.exit(main(argv))
             ^^^^^^^^^^
  File "mj_spec_remove_actuator.py", line 45, in main
    mj_model = spec.compile()  # this one fails.
               ^^^^^^^^^^^^^^
ValueError: Error: engine error: mj_initPlugin: plugin->init failed for plugin id 0
actuator not found for plugin instance 0
```

Workaround with raw xml:
```
    def replace_plugin_actuator_with_motor_actuator(match):
        name = match.group("name")
        joint = match.group("joint")
        forcerange = match.group("forcerange")

        gear = 1.0

        return f'<motor name="{name}" joint="{joint}" ctrlrange="{forcerange}" gear="{gear}" forcerange="{forcerange}"/>'

    xml_string = re.compile(
        r'<plugin\s+name="(?P<name>[^"]+)"\s+plugin="mujoco.pid"\s+instance="[^"]+"\s+ctrlrange="(?P<ctrlrange>[^"]+)"\s+forcerange="(?P<forcerange>[^"]+)"\s+dyntype="none"\s+joint="(?P<joint>[^"]+)"[^>]*\/>'
    ).sub(replace_plugin_actuator_with_motor_actuator, xml_string)

    xml_string = re.compile(
        r"\s+<extension>(?P<extension>.+?)</extension>",
        flags=re.DOTALL,
    ).sub("", xml_string)
```

quagla, 2024-09-19:
Hi @hartikainen, I assume that `instance=` lives inside the `actuator` block and not the `extension` block, right? If so, I will send a fix shortly.

hartikainen, 2024-09-22:
Hey @quagla! Thanks for the fast response on this one. I built the latest mujoco from source and testted this on the python side and it's still failing. Here's a python test case (which could be dropped directly into [specs_test.py](https://github.com/google-deepmind/mujoco/blob/91cedfd65c170af8df7b5baf25b08ea444c486d2/python/mujoco/specs_test.py#L565).):
```python
  def test_delete_plugin_actuator(self):
    spec = mujoco.MjSpec()
    spec.from_string(textwrap.dedent("""
      <mujoco model="MuJoCo Model">
        <extension>
          <plugin plugin="mujoco.pid">
            <instance name="pid1">
              <config key="kp" value="4.0"/>
            </instance>
          </plugin>
        </extension>

        <worldbody>
          <body pos="-0.06 0 0">
            <joint name="j1" type="slide" axis="0 0 1" />
            <geom size="1"/>
          </body>
        </worldbody>
        <actuator>
          <plugin joint="j1" plugin="mujoco.pid" instance="pid1" ctrlrange="0 0.15" />
        </actuator>
      </mujoco>
    """))

    for pluging_actuator in spec.actuators:
        pluging_actuator.delete()

    model = spec.compile()
```

This one fails with:
```
======================================================================
ERROR: test_delete_plugin_actuator (specs_test.SpecsTest.test_delete_plugin_actuator)
specs_test.SpecsTest.test_delete_plugin_actuator
----------------------------------------------------------------------
Traceback (most recent call last):
  File "mujoco/python/mujoco/specs_test.py", line 793, in test_delete_plugin_actuator
    model = spec.compile()
            ^^^^^^^^^^^^^^
ValueError: Error: engine error: mj_initPlugin: plugin->init failed for plugin id 0
actuator not found for plugin instance 0

----------------------------------------------------------------------
```

quagla, 2024-09-23:
Turns out this is not an issue with Python nor with mjSpec but with plugins :) let my try to fix it.

quagla, 2024-09-23:
Sorry, actually the error message above already tells you the issue. The PID plugin needs to have at least one actuator using it, so you can't delete all the actuators that use it. Note that there are two cases:
- The plugin instance is declared in `actuator`, in this case the actuator owns the plugin and deletes it when the actuator is deleted.
- The plugin instance is declared in `extension`, in this case the model (spec) owns the plugin, therefore when you delete an actuator, the instance still exists since another actuator might be using it. For the PID plugin, when no actuator is using the plugin, this results in an error. 

Do you think we should automatically delete all plugins that have no referencing object at model compilation or even when you delete all objects referencing it? @yuvaltassa WDYT?

quagla, 2024-09-23:
I think adding the support for directly doing the following is enough:

```
plugin = spec.plugins[0]
plugin.delete()
```

so I will send a fix soon.

hartikainen, 2024-09-23:
Yeah I think the safe thing to do here is to manually delete them with `plugin.delete()`, just like in your example. I tried that one too, but as you probably noticed, that one currently segfaults.

commit f4381e12 (2024-09-19): Delete plugins when their corresponding objects are deleted, if their instance belonged the object.

Fixes #2061.

PiperOrigin-RevId: 676402701
Change-Id: Ia4807d57a3763e2100702b7e7ec96c300d0a9cfb
commit 0ba11b96 (2024-09-23): Add `plugins` list to mjSpec Python API.

Fixes #2061.

PiperOrigin-RevId: 677762985
Change-Id: I8e407373e7b505d9b0f1c023e97c5ea92169bbf6
closed 2024-09-23 by copybara-service[bot]