google-deepmind/mujoco #2066: Make free joint possible within a body

HoangGiang93, 2024-09-19:
### The feature, motivation and pitch

I'm working on a parser to convert any format into another and would like the free joint can be used within a body. This can make the conversion into USD or other tree-based format better. Additionally, for example, consider I have a model of a house with many rooms and moving furniture for each room. I want to have each object as the child of each room to better maintain it. If this isn't possible, could you please explain why free joints must be at the top level?

yuvaltassa, 2024-09-20:
1. Are you using the `mjSpec` API? (you should)
2. The correct MuJoCo idiom is to not use bodies for static things (e.g. the rooms in your example) but [frames](https://mujoco.readthedocs.io/en/latest/XMLreference.html#frame-r), and to then [attach the frames](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mjs-attachframe) to the world body.

yuvaltassa, 2024-09-20:
Regarding you specific question, this is related to avoiding frame transformations. What you are asking is technically possible but would be less efficient. Your use case is precisely one of the reasons frames were introduced, since they are compile-time elements that incur zero run-time cost.

closed 2024-09-20 by yuvaltassa