google-deepmind/mujoco #2070: Using -DmjUSESINGLE with cmake has no effect

PeterOGB, 2024-09-20:
### Intro

New mujoco user here, I'm a retired software engineer with 40 years C/unix/linux experience.

I'm trying to  integrate mujoco into an existing application that runs on a Raspberry Pi5 and uses openGLES for rendering and gclm for maths.

### My setup

Raspberry Pi 5 running wayland on top of 64 bit PiOS (Debian 12).
cmake version 3.25.1
gcc (Debian 12.2.0-14) 12.2.0

### What's happening? What did you expect?

I want to build mujoco with mjtNum as float rather than double so as to more easily into existing appliacation.

I've tried adding -DmjUSESINGLE to the initial cmake command but it has no effect and cmake actually warns that "Manually-specified variables were not used by the project:    mjUSESINGLE".

I expected the resulting library to use floats, but it still uses doubles.

### Steps for reproduction

clone mujoco
cd mujoco
mkdir build && cd build
cmake .. -DCMAKE_INSTALL_PREFIX="/usr/local"   -DmjUSESINGLE=On

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

yuvaltassa, 2024-09-20:
I think you can add it e.g. [here](https://github.com/google-deepmind/mujoco/blob/78832db53f3134cda4373714c3e8489a7d9bc531/cmake/MujocoOptions.cmake#L88) ?

PeterOGB, 2024-09-20:
I did try adding it there but from memory it didn't seem to propagate to some of the dependencies.  I'll try it again.

PeterOGB, 2024-09-20:
No, it doesn't work...
[  9%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_collision_convex.c.o
/home/petero/mujoco/src/engine/engine_collision_convex.c: In function ‘mjc_penetration’:
/home/petero/mujoco/src/engine/engine_collision_convex.c:50:21: error: passing argument 1 of ‘mju_sub3’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   50 |         mju_sub3(dir->v, status.x1, status.x2);
      |                  ~~~^~~
      |                     |
      |                     ccd_real_t * {aka double *}
In file included from /home/petero/mujoco/src/engine/engine_collision_convex.c:27:
/home/petero/mujoco/src/engine/engine_util_blas.h:83:28: note: expected ‘mjtNum *’ {aka ‘float *’} but argument is of type ‘ccd_real_t *’ {aka ‘double *’}
   83 | MJAPI void mju_sub3(mjtNum res[3], const mjtNum vec1[3], const mjtNum vec2[3]);
      |                     ~~~~~~~^~~~~~

Please reopen !

yuvaltassa, 2024-09-20:
Okay, in that case this a different bug. I will rename and assign.

PeterOGB, 2024-09-20:
I don't think it is a doc issue.  libccd is not built separately, it's included in the mujoco tree, so I would have expected it to be reconfigured appropriately when using mjUSESINGLE.

PeterOGB, 2024-09-20:
My work-around fix was to change build/_deps/ccd-build/src/ccd/config.h after running cmake.

yuvaltassa, 2024-09-20:
If you know cmake well (I don't), do you know how to properly propagate `mjUSESINGLE` to libccd's `CCD_SINGLE` in the cmake config?

We'd very much appreciate a PR for this!

PeterOGB, 2024-09-21:
Configuring libccd can be done by changing the value set(ENABLE_DOUBLE_PRECISION ON) to set(ENABLE_DOUBLE_PRECISION OFF) in mujoco/cmake/MuJoCoDependencies.cmake  With this change (and setting mjUSESINGLE in top CMakeLists.txt) the library builds but the tests still don't build.

traversaro, 2024-12-07:
> Configuring libccd can be done by changing the value set(ENABLE_DOUBLE_PRECISION ON) to set(ENABLE_DOUBLE_PRECISION OFF) in mujoco/cmake/MuJoCoDependencies.cmake With this change (and setting mjUSESINGLE in top CMakeLists.txt) the library builds but the tests still don't build.

@PeterOGB an proposal to simplify this process is available at https://github.com/google-deepmind/mujoco/pull/2278, feel free to provide feedback if you like.