google-deepmind/mujoco #2072: `pip install mujoco` lacks sensor MJX feature updates from 3.2.3

nikashap, 2024-09-20:
### Intro

Hi there!

I am a graduate student at MIT, and I use MuJoCo for my research on reinforcement learning and human motor control.

### My setup

I've been using the python version of Mujoco in a conda environment successfully for a couple years now and recently transitioned to using MJX for Jax implementations. Here are the details of my environment and system:

`pip show mujoco` returns Version 3.2.3
API: Python
Architecture: Linux x86_64

### What's happening? What did you expect?

I updated my mujoco to 3.2.3 and expected to see sensor-related fields (nsensordata, name_sensoradr, etc.) associated with my mjx.Model instance, as I had defined `JOINTACTFRC` sensors in my xml file. The [Feature Parity](https://mujoco.readthedocs.io/en/latest/mjx.html#feature-parity) page says that this sensor type is now supported for MJX in 3.2.3.

Unexpectedly, I was getting errors that the sensor fields did not exist for both my mjx.Model and mjx.Data instances.

I dug into my python3.11/site-packages/mujoco/mjx/_src directory and noticed the sensor.py file (and other sensor-related files) were missing:

[image]

The 3.2.3 version of mjx/_src directory on github, however, contains a sensor.py file (and other sensor-related files) that were committed to main **before** the 3.2.3 tagged commit.

[image]

I triple-checked that I'm using Version 3.2.3, which I installed with `pip install --upgrade mujoco` 

### Steps for reproduction

1. Create a new conda or virtual environment
2. Run `pip install mujoco`
3. Check that the version is 3.2.3 with `pip show mujoco`
4. Navigate to the site-packages directory and view mujoco/mjx/_src/
5. Notice missing sensor.py files (and potentially other files)

### Confirmations

- [X] I searched the [latest documentation](https://mujoco.readthedocs.io/en/latest/overview.html) thoroughly before posting.
- [X] I searched previous [Issues](https://github.com/google-deepmind/mujoco/issues) and [Discussions](https://github.com/google-deepmind/mujoco/discussions), I am certain this has not been raised before.

erikfrey, 2024-09-20:
Hi @nikashap - thanks for the detailed documentation.  It helped me very quickly review your problem - I appreciate it!

I cannot repro this on my end.  When I run ```pip install mujoco-mjx```, I see that it installs `3.2.3` and I also see a `sensor.py` in the package contents.

I noticed that you are running `pip show mujoco` - can you also do `pip install --upgrade mujoco-mjx`?  These are two separate packages as they have different dependencies, but we keep them version-synchronized.

Closing but feel free to reopen if this doesn't fix your problem.

nikashap, 2024-09-20:
Hi @erikfrey - Wow, that exactly fixed my issue! The `pip install --upgrade mujoco-mjx` installs the correct `mujoco_mjx-3.2.3` distribution whereas just `pip install --upgrade mujoco` installs the earlier distribution version (`mujoco-mjx-3.2.0`). Thank you so much!